energia/Energia #817: Updating SLFS to latest version, add EMT

spirilis, 2016-01-05:
This will become a pull request for updating the SLFS library in all CC3100/CC3200 WiFi libraries with the latest version, deleting the old examples and adding the new examples.  As the API has changed a little bit (new functions, return code is no longer true/false but SL_FS_OK (0) for success and < 0 for error), it's best folks get used to the updated library rather than use the old one from pull request # 572.

rei-vilo, 2016-01-05:
Un-committing previous pull request might be safer.

Also, EMT has a dedicated `fatfs` library.

spirilis, 2016-01-05:
Nah I figured out the deltas.  Pullreq forthcoming...

spirilis, 2016-01-05:
Pullreq as #819

rei-vilo, 2016-01-19:
There's a nice feature available on the Arduino IDE for the Arduino Yun. If the folder of the folder has a sub-folder called `www`, it is automatically uploaded to the board. 

For more information, please refer to [Arduino Yún · Intro to web server](http://scuola.arduino.cc/lesson/b4EoRkV/Arduino_Yn_Intro_to_web_server).

rei-vilo, 2016-01-19:
New ticket #827.

robertinant, 2016-01-19:
This would require enhancing cc3200prog and probably the IDE to instruct cc3200prog to upload the www folder. I'll put it on the list of enhancements for cc3200prog.

spirilis, 2016-01-19:
Another option would be an Energia "feature" to auto-generate a sketch that includes the files in a C array, and produces SLFS code to upload them, then compiles, uploads and runs.  I think the cc3200prog route is better.

closed 2016-02-11 by spirilis