energia/Energia #820: Solution Rate per Year

rei-vilo, 2016-01-05:
Here's the solution rate per year for the tickets.

[image]

spirilis, 2016-01-05:
Somewhat along the lines of a rant I posted on 43oh not long ago, I am curious who all is working on / contributing to / processing all the details, looking for bugs, etc. with this project.  All I really know of is Robert, rei-vilo, Stefan and a few TI'ers like Trey doing C2000 work, myself sporadically but not much in the past year, but who else is really looking at it?  I can imagine the user base is fairly significant especially with TI's university work, just wondering what the support base looks like.

rei-vilo, 2016-01-06:
Same stats for pull requests.

[image]

Overall, the results are impressive!