energia/Energia #821: OneWire library

ozarchie, 2016-01-08:
Is this code in OneWire .cpp correct?
Should it be-:
 pinMode(pin, OUTPUT);
 digitalWrite(pin, HIGH);

// Perform the onewire reset function.  We will wait up to 250uS for
// the bus to come high, if it doesn't then it is broken or shorted
// and we return a 0;
//
// Returns 1 if a device asserted a presence pulse, 0 otherwise.
//
uint8_t OneWire::reset(void)
{
	pinMode(3, OUTPUT);
	digitalWrite(3, HIGH);

ozarchie, 2016-01-08:
There is also this thread from TI saying that it may require extra config to turn the port around -:https://e2e.ti.com/support/wireless_connectivity/simplelink_wifi_cc31xx_cc32xx/f/968/t/456513

spirilis, 2016-01-08:
Ah, Open Drain mode.  Probably the right thing to do, but not something Energia supports natively IIRC.

ozarchie, 2016-01-08:
Any idea about the 'pin' vs '3' in OneWire.cpp ?

spirilis, 2016-01-08:
No idea, never actually interfaced a TI MCU with a 1-wire part (well, I did check out a 1-wire temp+humidity sensor one time, but I was using a G2553 with straight C when I did that, no energia).

ozarchie, 2016-01-08:
I  changed it to 'pin' to get some DS1280s to work.
Not sure if there was a problem. But it didn't look right

rei-vilo, 2016-01-08:
I used the [OneWire](https://github.com/energia/Energia/tree/master/libraries/OneWire) library successfully on the MSP430G2553 and Energia with the [DS2401 Silicon Serial Number](https://www.maximintegrated.com/en/products/digital/memory-products/DS2401.html).

rei-vilo, 2016-01-08:
Check lines 124 and 125. Hard-coded pin `3` doesn't very clean...

```
	pinMode(3, OUTPUT);
	digitalWrite(3, HIGH);
```

robertinant, 2016-01-08:
Thanks for the report @ozarchie!

This was left over code from when I was debugging the library. This code is dead and erroneous. I just removed it and pushed the result: 6dce194f61e7b5d727cce52ff470ad41ec689a68

closed 2016-01-08 by robertinant

ozarchie, 2016-01-09:
Thanks - I have done the same and the DS12B80s are working well - currently three on the Onewire bus, and being compared against the tmp006 using the Adafruit library. Thanks for your response.