energia/Energia #822: CC3200: Inconsistent Resolutions Between Non-EMT and EMT

rei-vilo, 2016-01-10:
The very same sketch with `analogRead()` throws different values. 

Actually, they have different resolutions: **non-EMT** works on **12 bits** (maximum = `4095`) while **EMT** sticks with **10 bits** (max = `1023`).

* Analog Input CC3200 non-EMT

>
2294
2293
2294
2293
>

 * Analog Input CC3200 EMT
>
573
573
573
573
>

* Sketch

```
void setup()
{
  Serial.begin(9600);
  delay(100);
  Serial.println("pin 6 = Analog Input");
  pinMode(6, INPUT);
  for (uint8_t i = 4; i > 0; i--)
  {
    Serial.println(analogRead(6), DEC);
    delay(500);
  }
}

void loop()
{

}
```

race2fly, 2016-01-15:
这程序是Arduino的吧
CC3200程序有什么问题

rei-vilo, 2016-01-15:
@race2fly 
Please post in English so everyone can understand. Thank you.