energia/Energia #823: Issues with Ubuntu 15.10 and msp430g2 launchpad

cjdg, 2016-01-11:
I have issues working with the serial ports in linux, nothing works when i put the micro to sent serial data

rei-vilo, 2016-01-11:
Could you please be more specific?

* Linux distribution
* Energia release
* LaunchPad reference and version
* USB connection —direct or [using hub](https://github.com/energia/Energia/wiki/Using-hub)

cjdg, 2016-01-11:
Ubuntu 15.10 64 bit
Kernel version 4.2.0
Energia 17
msp430g2 1.5
As far as i know is an old issue with the kernel, with cdc_acm, which seems to dont get enough pressure to fix this, i was using patches, but after the last upgrade stop working.

spirilis, 2016-01-11:
I've wrestled with this before, the main thing I had to keep in mind is DO NOT send any data from MCU -> Linux PC until you have a terminal app (or Serial Monitor) window open.  If you do, unplug the LaunchPad and plug it back in, then hold the RESET button until you get the serial monitor or terminal window open, then release RESET.

Alternately, organize your sketches to all check the P1.3 pushbutton in setup() before proceeding to initialize the Serial et al.

cjdg, 2016-01-12:
Well have good news, the thing works like charm with raspberry pi b (first edition) with debian Jessie and kernel 4.1.13

closed 2016-01-12 by rei-vilo