energia/Energia #825: analogRead with WIFI

tae-jun, 2016-01-18:
My Environments:
* RedBearLab WiFi Mini
* Mac OSX
* Trying to read current value using ACS712 (of course used to voltage divider to lower output voltage)

Everything works fine, but when I use `WiFi.begin(ssid, password)` problem arises.

Value returns from `analogRead()` is **incorrect**

I found that this issue occurred [here] (http://forum.43oh.com/topic/5878-cc3200-wifi-and-analogread/) already.

According to link above,
``` Arduino
analogRead(2);  // discard
analogRead(2);  // discard
analogRead(2);  // discard
analogRead(2);  // discard
int sensorValue = analogRead(2);  // Read ADC
```
This code will be helpful.

But actually, still quarter of return value of `analogRead()` is incorrect.

Is there any solution?

rei-vilo, 2016-01-18:
Have you contacted RedBear?

tae-jun, 2016-01-19:
I just contacted now :-)