energia/Energia #826: Clicking the menu 'Tools' freezes the whole IDE

engiecat, 2016-01-19:
Hello.

I am a user who recently returned to use Energia IDE.
However, one very critical problem renders the IDE unusable - clicking on the menu 'Tools' freezes the whole IDE for 1+ minutes, preventing me to select COM port and board type.

The version used is 0101E0017.
The OS I am using is Windows 10 Pro, and I am using i5-2467M CPU.
Arduino IDE is working very well on this machine.

rei-vilo, 2016-01-19:
The same issue used to happen with Arduino 1.0 Energia is based on. 

The delay is caused by the enumeration of all serial ports, including the Bluetooth ones. It could take up to 5 minutes. 

As a Window user, give [Visual Micro](http://www.visualmicro.com) a try. It's a free plug-in for Visual Studio and supports Energia.

robertinant, 2016-01-19:
The next release of Energia will be based on the latest and greatest Arduino IDE and this issue will be solved by moving to that new IDE. For this reason, it does not make sense for us to fix it in the current IDE.

closed 2016-01-19 by robertinant

rei-vilo, 2016-01-19:
See Update the IDE to release 1.6.7 of Arduino #588.