energia/Energia #827: Upload Website to CC3200 and Use SLFS to Read Them

rei-vilo, 2016-01-19:
There's a nice feature available on the Arduino IDE for the Arduino Yun. If the folder of the folder has a sub-folder called `www`, it is automatically uploaded to the board. 

Files are then read using the SLFS library #817.

For more information, please refer to [Arduino Yún · Intro to web server](http://scuola.arduino.cc/lesson/b4EoRkV/Arduino_Yn_Intro_to_web_server).

leedennis519, 2016-02-24:
Hi, 
Is this tested and working ? will the AP provisioning example for cc3200 grab that website instead of the demo website ?

dx266, 2016-03-10:
hi
So apparently from what I have red a similar thing is not available on energia yet. So is there a way to upload a website on the flash using energia for a wifi mini board? thank you