energia/Energia #829: Major changes between Energia 15 and 17

TransientResponse, 2016-01-19:
I have a sizeable commercial firmware project using Energia and the TM4C1294NCDPT, and upon trying to update from Energia 15 to Energia 17 results in a lot of weird behavior on the device. The firmware builds just fine with no errors or warnings (besides the usual ones in Ethernet.h and related files), but does strange things on the device. Were pin defines for that part changed between 15 and 17? The compiler version is the same between both of them, so I know it's not a GCC issue.

rei-vilo, 2016-01-19:
Which libraries are involved in your project?

TransientResponse, 2016-01-19:
Ethernet (UDP and TCP), Wire, EEPROM, TivaWare driverlib timer, sysctl, eeprom, and shamd5. The others are a few custom ones: a DS2431 (Maxim OneWire EEPROM) library written before Energia had OneWire, and an MCP4728 library copy-and-pasted from Arduino. 

I don't think it's the DS2431 library as it appears to be reading that data from the external EEPROM correctly, it's rather how the device responds to the commands issued over UDP that's the problem.

rei-vilo, 2016-01-19:
Some hints:

* Test each library implied on your project separately.
* Compare the libraries between releases 15 and 17 of Energia to spot the differences.

robertinant, 2016-01-19:
Can you please describe the weird behavior that you are seeing?

robertinant, 2016-01-19:
I just did a git log for TivaC to see if I can spot something obvious.
- driverlib did not change since 2014.
- Wire changed it's default module from 1 to 3. If you are using Wire.setModule() in your code then this does not apply. It seems that Wire is still working for you so probably not the issue.
- EEPROM was enhanced with .update() but has no impact on the previous implementation.

The only changes to ethernet where a couple of enhancements:
- https://github.com/energia/Energia/commit/116452b173bb6ad029d1edbf600f07af230200dd
- https://github.com/energia/Energia/commit/827c53184c4698ff54a002174b07dfe15119d6d2

I looked through these changes and can not spot something obvious that would cause the issues you are seeing.

robertinant, 2016-01-19:
Can you please try to replace the Ethernet library in Energia 17 with the one in Energia 15 to figure out if the Ethernet library could be the cause?

TransientResponse, 2016-01-19:
Certainly. 

We're building a three-channel industrial lighting controller, using the MCP4728 multi-channel DAC to drive the CTRL pins on an LT3797. As I send a command to set the brightness (current) on different channels, the firmware sends responses confirming the correct input and channel but the hardware does not act accordingly. Setting Channel 1 does nothing, setting Channel 2 influences channel 1 hardware, etc. 

The MCP4728 uses Wire, but I'm manually calling setModule(0) to make that work. 

I will try replacing the Ethernet files as you suggested.

TransientResponse, 2016-01-19:
No effect after replacing the Ethernet files. 

I'll try writing a short sketch that just uses the MCP4728 to set a value to the DAC and see what happens.

TransientResponse, 2016-01-19:
It would appear that it's the MCP4728 DAC that's the culprit. Source code for the library is hard to get to, but I can provide it. 

Also, copying over E15's Wire library does not fix it.

robertinant, 2016-01-20:
Did anything change code wise for the MCP4728?

TransientResponse, 2016-01-21:
No. I didn't change any of my code, only updated Energia. I'll see if I can put a logic analyzer on the MCP's I2C lines to see any difference between E15 and E17.

itiserik, 2016-01-21:
Did you apply pr #799 ?
setmodule(0) -> the hardware i2c...

TransientResponse, 2016-01-21:
I will apply that patch tomorrow when I'm back at work.

StefanSch, 2016-01-22:
Patch #799 applies for the MSP430 devices, this issues is based on TivaC, based on the comment above.

TransientResponse, 2016-01-22:
Didn't even notice that it was only for MSP. Well the problem still has something to do with I2C, and copying E15's Wire library over to E17 does not fix it.