energia/Energia #830: Enhance Wire to allow for multiple instances

robertinant, 2016-01-20:
With the current implementation of Wire (I2C) only one instance is possible due to some none instance based variables in the I2C state machine.

itiserik, 2016-01-20:
These multiple instances are only needed when the hardware supports multiple i2c interfaces. It is only for devices with both USCI_B0 and USCI_B1. Just because a software i2c interface is integrated in twi.c/h, I assume a need to have multiple instances. But I think we better go with a true software Wire lib.

And before inventing something new on this lib please apply some pull requests (otherwise they can't be applied anymore), fix master repeated start and cleanup the code. Please test before releasing a new energia version.

rei-vilo, 2016-01-20:
Feel free to join the team! You're welcome!

altineller, 2017-09-25:
I am looking for ways to use multiple i2c ports on the TM4C123. I read the Wire.h and Wire.cpp that belongs to the TivaC hardware. And I noticed at the end of Wire.cpp

void TwoWire::setModule(unsigned long _i2cModule)
{
    i2cModule = _i2cModule;
    if(slaveAddress != 0) begin(slaveAddress);
    else begin();
}

//Preinstantiate Object
TwoWire Wire;

1. that every time we setModule, it will start that module again with begin()
2. object is preinstantiated

There is this structure: pin and device configuration is laid out in arrays. every time a function executes it executes according to i2cModule number.

So it should work without the begin in setModule statement? Maybe there is a reason why it is made like this?

keithwm422, 2020-03-16:
I tried a simple test without the setModule having a begin and was not able to successfully communicate with multiple devices using the Wire lib. I am using the TM4C123 and am working on adjustments to the lib to use mutliple instances. Are there any updates here on this (the issue is still open as it appears to me).