energia/Energia #832: Energia 17 fails to launch on Windows 10 (Surface Book)

dspdon, 2016-01-22:
Energia fails to launch on an MS Surface Book running Windows 10.  The splash screen appears, but goes away and the process dies thereafter.  Note that it's launching just fine on a Lenovo W530 laptop with Win7, same download (Win 64b).  Likely different JDK's, but now I'm just guessing as to reasons...

rei-vilo, 2016-02-05:
Have you checked the Java version?

dspdon, 2016-02-05:
I didn't see a spec for Java version to use/not use when running Energia ... but I frequently miss stuff in front of my nose.  ;-)  I installed the latest at the time, JVM v8u71.  What's needed?

rei-vilo, 2016-02-05:
Energia relies on Java 6, which is a rather antiquated version.
 
See #832 Energia 17 fails to launch on Windows 10 (Surface Book), #728 Java Release Issue With Mac OS X 10.11 El Capitan, #533 Readiness for Java SE 8.
 
Energia is on the process of being ported to Arduino 1.6.7. Stay tuned!
 
See #588 Update the IDE to release 1.6.7 of Arduino.

dspdon, 2016-02-05:
Exactly what I needed to learn - thank you!

rei-vilo, 2016-04-27:
Update the IDE to release 1.6.7 of Arduino #588 may solve this issue.