energia/Energia #833: attachInterrupt example in *shipped* reference still needs setup() fix

petesoper, 2016-01-23:
```
  /* Enable internal pullup. 
   * Without the pin will float and the example will not work */
  pinMode(PUSH2, INPUT_PULLUP);
```

Copied from the two year old forum thread at
  http://forum.43oh.com/topic/4955-correction-attachinterrupt/

rei-vilo, 2016-01-23:
Please provide more information: board, MCU, Energia release, ... Thank you.

petesoper, 2016-01-23:
Not relevant. Start any Energia, go to help/reference, click on "attachInterrupt", scroll down to example code. These two lines need the code above inserted between them:
```
 digitalWrite(GREEN_LED, state);
  attachInterrupt(PUSH2, blink, FALLING); // Interrupt is fired whenever button is pressed
```
In my opinion it is a MAJOR BUG to have your shipped(downloaded/installed package) docs not be derived from the same files that generate the online docs or visa versa. How many deltas are in these two sets of docs???
That said, let me take this opportunity to applaud the Energia development team for all your hard work!

robertinant, 2016-01-25:
@petesoper thanks for the report. I thought about this and from a h/w point of view the example is correct. The assumption is that an external pull-up resistor is provided. I think the real issue is that the documentation does not mention a pull-up at all. What I will do is to fix the documentation mentioning an external/internal pull-up and then also add the pinMode(<pin>, INPUT_PULLUP); call to the Sketch but comment out with the comment that it should be uncommented IF there is no external pull-up.

Robert

petesoper, 2016-01-25:
> pinMode(, INPUT_PULLUP); call to the Sketch but comment out with the

Hello Robert,
    Thanks for the quick response. The bug is that the example does not 
match the corresponding one on the Energia web site at this URL:
http://energia.nu/reference/attachinterrupt/

    Note the comment on 20 January, 2014 by "@Mention" on this page:

http://forum.43oh.com/topic/4955-correction-attachinterrupt/ 
<http://forum.43oh.com/topic/4955-correction-attachinterrupt/>

"It was corrected on energia.nu but the reference page that ships with 
Energia has not been corrected. This page was probably missed when 
syncing things. Will get it synced for the next release."

This obviously didn't happen. This current #833 is a request for this 
fix to get applied.

I have to take issue with the notion that "from the h/w point of view 
the example is correct". Without the pinMode call the hardware isn't 
going to do anything in response to changes of potential on the pin, and 
this is independent of the subject of pullups.

Regards,
Pete

robertinant, 2016-01-25:
I read through it and looked through the examples. Currently there is no attachtInterrupt example shipped  for the basic examples in Energia. Am I missing something here? Which example are you referring to that is shipped in Energia?

petesoper, 2016-01-25:
Between lines 84 and 85 of this file there need to be the added lines.
  https://github.com/energia/Energia/blob/master/reference/AttachInterrupt.html

This will make the example code above match the example code on this page:
    http://energia.nu/reference/attachinterrupt/

   
  

That's the bug.

robertinant, 2016-01-25:
Thanks for pointing that out! Now I get it. The reference shipped does not match the doc on energia.nu. Will fix it ASAP.

petesoper, 2016-01-25:
And I realize what I should have put into the issue from the get go. Sorry for the confusion.