energia/Energia #835: Request for soft serial and parity in MSP432

Suyogaerospace, 2016-01-31:
Hi
I think all four serial ports should be made available for msp432 with parity support and if possible soft serial too.

dronus, 2016-06-28:
Actually the original Arduino variants have this settings now, see:  https://www.arduino.cc/en/Serial/Begin
So Energia should just implement the updated API.

This is especially important for the MSP432 boards, as the settings can't be changed by low level poking like it is possible on the Arduino and Energia MSP430 platforms, as it is implemented by precompiled runtime code on the MSP432.

RickKimball, 2016-06-28:
You might look at the change I made for the msp430 with usci. The uart on the msp432 is pretty close to this. You just need to implement the appropriate arm code.

http://forum.43oh.com/topic/9063-usci-patch-for-parity/

dronus, 2016-07-14:
The UART is controlled by TI runtime libraries on the MSP432 EMT core, so this is a completely different thing. No registers are set, but TI libraries linked and configured by config struct's. 

Currently these structs are only configured by the baudrate setting of Serial.begin, and this initalisation code is buried in Wiring runtime library binary blob. 

So to "implement" those settings, one have to change the Serial.begin source code and rebuild the binary libraries. However, there seems to be no documentation at all how those where built.

closed 2016-08-29 by robertinant