energia/Energia #836: Provide Other Ports on LaunchPads

rei-vilo, 2016-01-31:
See #298 and #835.

rei-vilo, 2016-02-07:
The F28069 has two UART ports: one on pins 3 / 4, and another on pins 43 / 44, but it looks like Energia only uses one, `Serial`. Please implement the second. Thank you.

[image]

spirilis, 2016-02-08:
@rei-vilo side note, is that F28069 pin map diagram on energia.nu yet?  couldn't find it (someone on IRC #43oh asking about using a bluetooth module with the F28069)

rei-vilo, 2016-02-08:
@spirilis Not yet, but I sent it a while ago. 

Actually, the post was prompted by this thread at 43oh http://forum.43oh.com/topic/9347-bluetooth-module/

spirilis, 2016-02-08:
Ah right... think it's the same person who was asking about this on IRC.  I nudged him to the forum instead.