energia/Energia #837: SPI on energia, documentation, begin-/endTransaction

baettigp, 2016-01-31:
Hi, I am trying to port a library (for the ADS1258) over from an Arduino uno to a MSP430F5529 as I am running out of resources (RAM, and in a next step pins).
I am trying to get the SPI side working now and noticed that on the energia website many of the spi documentation pages still seem to reference the arduini (uno, due, mega) instead of the MSP variants. http://energia.nu/reference/spi/spi_setclockdivider/ does unfortunately not tell me how to set the spi clock I need (4 MHz) with the 25 MHz F5529.
Also, is it planned to implement the 
SPI.beginTransaction(SPISettings(SPI_SPEED, xSBFIRST, SPI_MODEy));
and SPI.endTransaction functions into energia?

rei-vilo, 2016-01-31:
Try with the already implemented functions `setBitOrder()`, `setClockDivider()` and `setDataMode()` available on the [SPI Library](http://energia.nu/reference/spi/).

spirilis, 2016-02-01:
Definitely a good question about implementing SPI.beginTransaction, endTransaction and the SPISettings class.  That is a Paul Stoffregen invention that has since become a standardized part of the Arduino SPI API.  In particular the SPISettings concept makes SPI clock speed setting a no-brainer (no guessing what the clock is "dividing" off of, and the TI ARM Cortex-M devices with ROM driverlib support specifying SPI clock speeds based on a uint32_t clock speed parameter which should make implementation relatively simple).

The most hairy part of the SPI Transaction support API will be the .usingInterrupt() function and its ultimate implementation.  This requires SPI having hooks into the platform's native interrupt request mechanism and being able to selectively disable/re-enable port interrupts as necessary, or doing a general interrupt disable.

Topic for another thread and another day though.

robertinant, 2016-02-03:
@baettigp the default clock by default is kept as close as possible to ~4 MHz. In the case of the F5529 running at 25MHz it is 25000000/4000000. If the remainder is > 0, 1 is added to keep the clock speed under 4 MHz. So in this case
25000000/4000000 = 6.25. The remainder is > 0 so the divider becomes 7. 
25000000/7 = 3571428 Hz for the clock speed. If you are OK with it being a bit above 4 MHz then you can use SPI.setClockDivider(6) which will get you 4166666Hz.

The SPI documentation indeed needs work. I'll see if I can get to that later this week.

As for the beginTransaction() API. It has been on the list for a while. What has been keeping us from implementing it is that we have to do it across the board for all devices supported by Energia and quite a bit of work. We probably won't be able to get to it before the next release but hopefully the release after that.

robertinant, 2016-08-29:
This issue was moved to energia/msp430-lg-core#17

closed 2016-08-29 by robertinant