energia/Energia #838: Serial Monitor issue on Debian Wheezy KDE Linux

S4rdus, 2016-01-31:
Hello everyone,
I mounted a  MSP430G2553 on my Linux machine, i can to upload sketchs normally, "update programmer" works normally also:
I Trying to open interface 1 on 010
Initializing FET...
FET protocol version is 30394216
Configured for Spy-Bi-Wire
Set Vcc: 3000 mV
Device ID: 0x2553
  Code start address: 0xc000
  Code size         : 16384 byte = 16 kb
  RAM  start address: 0x200
  RAM  end   address: 0x3ff
  RAM  size         : 512 byte = 0 kb
Device: MSP430G2553/G2403
Number of breakpoints: 2
Chip ID data: 25 53

However when i enable the "Serial Monitor" i get this: 

processing.app.SerialNotFoundException: Serial port '/dev/ttyACM0' not found. Did you select the right one from the Tools > Serial Port menu?
	at processing.app.Serial.<init>(Serial.java:165)
	at processing.app.Serial.<init>(Serial.java:92)
	at processing.app.SerialMonitor.openSerialPort(SerialMonitor.java:209)
	at processing.app.Editor.handleSerial(Editor.java:2630)
	at processing.app.Editor$20.actionPerformed(Editor.java:777)
	at javax.swing.AbstractButton.fireActionPerformed(AbstractButton.java:2022)
	at javax.swing.AbstractButton$Handler.actionPerformed(AbstractButton.java:2346)
	at javax.swing.DefaultButtonModel.fireActionPerformed(DefaultButtonModel.java:402)
	at javax.swing.DefaultButtonModel.setPressed(DefaultButtonModel.java:259)
	at javax.swing.AbstractButton.doClick(AbstractButton.java:376)
	at javax.swing.plaf.basic.BasicMenuItemUI.doClick(BasicMenuItemUI.java:833)
	at javax.swing.plaf.basic.BasicMenuItemUI$Handler.mouseReleased(BasicMenuItemUI.java:877)
	at java.awt.Component.processMouseEvent(Component.java:6525)
	at javax.swing.JComponent.processMouseEvent(JComponent.java:3324)
	at java.awt.Component.processEvent(Component.java:6290)
	at java.awt.Container.processEvent(Container.java:2234)
	at java.awt.Component.dispatchEventImpl(Component.java:4881)
	at java.awt.Container.dispatchEventImpl(Container.java:2292)
	at java.awt.Component.dispatchEvent(Component.java:4703)
	at java.awt.LightweightDispatcher.retargetMouseEvent(Container.java:4898)
	at java.awt.LightweightDispatcher.processMouseEvent(Container.java:4533)
	at java.awt.LightweightDispatcher.dispatchEvent(Container.java:4462)
	at java.awt.Container.dispatchEventImpl(Container.java:2278)
	at java.awt.Window.dispatchEventImpl(Window.java:2750)
	at java.awt.Component.dispatchEvent(Component.java:4703)
	at java.awt.EventQueue.dispatchEventImpl(EventQueue.java:758)
	at java.awt.EventQueue.access$500(EventQueue.java:97)
	at java.awt.EventQueue$3.run(EventQueue.java:709)
	at java.awt.EventQueue$3.run(EventQueue.java:703)
	at java.security.AccessController.doPrivileged(Native Method)
	at java.security.ProtectionDomain$1.doIntersectionPrivilege(ProtectionDomain.java:75)
	at java.security.ProtectionDomain$1.doIntersectionPrivilege(ProtectionDomain.java:86)
	at java.awt.EventQueue$4.run(EventQueue.java:731)
	at java.awt.EventQueue$4.run(EventQueue.java:729)
	at java.security.AccessController.doPrivileged(Native Method)
	at java.security.ProtectionDomain$1.doIntersectionPrivilege(ProtectionDomain.java:75)
	at java.awt.EventQueue.dispatchEvent(EventQueue.java:728)
	at java.awt.EventDispatchThread.pumpOneEventForFilters(EventDispatchThread.java:201)
	at java.awt.EventDispatchThread.pumpEventsForFilter(EventDispatchThread.java:116)
	at java.awt.EventDispatchThread.pumpEventsForHierarchy(EventDispatchThread.java:105)
	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:101)
	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:93)
	at java.awt.EventDispatchThread.run(EventDispatchThread.java:82)

"Serial Port" is often grey.
The LaunchBoard is correctly wired..
Could someone help me? 
Thank you.. 
MatH

robertinant, 2016-02-03:
The programmer on the G2 LaunchPad is a composite USB device (CDC/HID). On Linux it can take quite a bit of time for the kernel to enumerate the device. Hence you might have to wait some time for the port to show up in the Serial port menu. Ones it does, you should be able to open the serial monitor.

The "Serial port '/dev/ttyACM0' not found" is a typical case of the LaunchPad not yet enumerated. The only solution is to wait for the port to show up and only then open the Serial monitor.

closed 2016-02-03 by robertinant

S4rdus, 2016-02-04:
Solved!
Thanks