energia/Energia #839: TM4C1294 Ethernet randomly hangs

lcafaro, 2016-02-01:
Hello,

this simple code is showing random behavior. Data is only sent 2-3 times, sometimes up to a dozen times, then it either completely stops or send it all out instantly, like a buffer being flushed.

```
#include <Ethernet.h>

void setup(void) {
  Serial.begin(9600);
  Ethernet.begin(0);
  Serial.println(Ethernet.localIP());
}

void loop(void) {
  EthernetClient client;
  client.connect("192.168.1.219", 8080);
  client.write("abc\n\n");
  client.stop();
  sleep(1000);
}
```

I'm listening on the PC side with `nc -k -l 8080`.

Any hints? Am I doing something wrong? client.stop() / client.flush() don't seem to help. Neither do playing around with the object (instantiating it only it once, setting up a single connection, using print() instead of write(), passing data byte by byte... whatever).

Hardware is TM4C1294 / EK-TM4C1294XL.

Thanks

[crossposted to 43oh forums]

rei-vilo, 2016-02-01:
Have a look at the E2E forum http://e2e.ti.com. 

A similar issue was reported but I can't remember the answer.

lcafaro, 2016-02-01:
Hello,
I've checked but the only similar issue I've found was fixed in #447.
Thanks