energia/Energia #840: TM4C1294 Ethernet + ADC conflict

lcafaro, 2016-02-01:
Hello,
Ethernet and ADC seem to conflict on TM4C1294. Sample code:
```
#include <Ethernet.h>

EthernetClient client;
int sensorPin = A3;
float sensorValue = 0;

void setup() {
  Serial.begin(9600);
  delay(100);
  Ethernet.begin(0);
}

void loop() {
  sensorValue = analogRead(sensorPin);
  Serial.println(sensorValue);
  delay(300);
}
```

The ADC values seem to jump around randomly when Ethernet.begin(0) is called.

By "jump around" I mean this: http://pastebin.com/WDLhm0v0 - as you can see,  there's way more variance in the second column, which is samples taken after Ethernet.begin(0);

More precisely, analog samples with Ethernet disabled range from 265 to 287 (acceptable, within specs of the analog sensor I'm using) while samples with Ethernet enabled range from 127 to 368 (way off).

A simpler way to reproduce is to ground the analog input pin and check the samples. With Ethernet disable they range from 0 to 8, but as soon as Ethernet is enable they range from 0 to 47 in my tests.

I've tried to use other analog input pins and I get the same result.

Hardware is a TM4C1294 on the EK-TM4C1294XL, aka "Connected Launchpad".

Any ideas?

Thanks

[crossposted on 43oh forums]

GNZFB, 2016-05-16:
Same problem here. No ethernet, no issues, but as soon as I use Ethernet.begin(0) the ADC readings go bananas. My project is a shitty oscilloscope so let me show you some illustrated examples (of course same signal, same conditions with the only difference of enabling ethernet as said):
[image]
[image]

rei-vilo, 2016-05-16:
Some investigation is needed, as the culprit could be 

* the interference between Ethernet and ADC,
* Energia implementation.

Could you please try a similar program with CCS without using the Energia framework?

tthtlc, 2019-10-28:
The internal diagram of TM4C1294XL are as follows:

[image]

and you can see that ethernet and other I/O are all on the same bus:  ie, it is not possible to have concurrent processing of ethernet I/O and other I/O.   That may explain why your graphics display dropped in resolution.