energia/Energia #841: wiring_digital.c missing for MSP432 Launchpad?

Terstegge, 2016-02-02:
I am debugging an Energia sketch in CCS. When I try to step into digitalWrite(), the debugger is looking for the file wiring_digital.c, which is missing in Energia 17.
Older posts say there should be a file 
<energia_home>/hardware/msp432/cores/msp432/wiring_digital.c
but I can not find it ..

rei-vilo, 2016-02-02:
Have a look for `wiring_digital.c` under `Energia.app/Contents/Resources/Java/hardware/emt/ti/runtime/wiring` for Mac OS X or similar for other OSes.

Terstegge, 2016-02-02:
Thanks for the quick answer, but I don't seem to get the point (sorry). I can not locate the path you stated in my Energia 17 installation. The only wiring_digital.c files I find are 
./hardware/msp430/cores/msp430/wiring_digital.c
./hardware/emt/ti/runtime/wiring/wiring_digital.c
./hardware/lm4f/cores/lm4f/wiring_digital.c
./hardware/c2000/cores/c2000/wiring_digital.c
./hardware/cc3200/cores/cc3200/wiring_digital.c

rei-vilo, 2016-02-02:
Which release of Energia and which OS are you using? 

On Mac OS X, the `wiring_digital.c` is available for the MSP432 on the folder I've mentioned earlier.

Please note CCS requires Energia 16 and doesn't support Energia 17 for the moment.

Terstegge, 2016-02-02:
I am using Windows10 in a VirtualBox, running CCS v4 and Energia 17 (0101E0017 as mentioned above). So far I had no problems at all in integrating Energia 17 into CCS v6. But of course debugging is not available in Energia, so missing source files might not be a problem for this IDE. But with CCS, it would be nice to debug into the Energia libraries, also showing the source code of those libraries. I have done this before with MSP430 Launchpads, but with my MSP432 Launchpad (which will be used in university classes) the related sources are not existing ...

rei-vilo, 2016-02-02:
The RTOS library comes pre-compiled on an archive. I guess an include directory is missing on the project settings.

There are many alternative solutions:

- All platforms: CCS Cloud 
- Mac OS X: CCS for Mac, although I don't recommend it
- Mac OS X: [embedXcode plug-in for Xcode](http://embedXcode.weebly.com), leveraging the Energia and Energia MT frameworks and all GNU standard tools likes GDB and Doxygen.
- Windows: [Visual Micro plug-in for Visual Studio](http://visualmicro.com), with software emulated debugging.

Terstegge, 2016-02-02:
Ah, okay. So there is no source code available for the MSP432, because of the sourceless RTOS MT library provided by TI. No problem, me and my students will then stick to CCS (or CCS Cloud) for SW exercises etc. Thanks again for the quick help!

rei-vilo, 2016-02-02:
I had a look at a Windows-based computer. 

`wiring_digital.c ` does exist under `energia-0101E0017\hardware\emt\ti\runtime\wiring\wiring_digital.c`.

closed 2016-02-02 by rei-vilo

Terstegge, 2016-02-02:
Great, thanks!

robertinant, 2016-02-02:
As @rei-vilo mentioned, the sources for the emt (Energia Multi Tasking) part of MSP432 support are available. When debugging in CCS and stepping int digitalWrite(), it should give you an option to "find the file". There is also an option in CCS to add source search paths.

Terstegge, 2016-02-02:
Thanks, I have used the "find the file" option already with success! And thanks for explaining the 'emt' abbreviation - I would have looked into that directory if I had known the meaning!