energia/Energia #842: SSL Support for Ethernet

dakotaewigman, 2016-02-03:
It looks like WIFI has the ability to connect to connect via https, could you add that for Ethernet?

spirilis, 2016-02-03:
Not simple at all.  The WiFi leverages the fact that the CC3100/CC3200 WiFi Network Processor has SSL built in (proprietary TI firmware at play there).  For TM4C1294 we'd have to kludge some open source SSL library into place to cover that (and the TM4C MCU would be expending CPU cycles to process it).  Also not sure how the TM4C1294 vs new TM4C129E chips play into this... the 129E has hardware crypto engine, is there an open source SSL library that can use it?

robertinant, 2016-02-03:
WolfSSL (CyaSSL) supports lwIP and I am told that we already have a port of the WolfSSL for the TM4C129E. Not sure if WolfSSL supports a emulated crypto engine to also enable it on the TM4C129.
I do not however have any plans to add support for this in the foreseeable future.

closed 2016-02-03 by robertinant