energia/Energia #843: MSP432P401R hangs when Wire.endTransmission() is executed

engiecat, 2016-02-03:
Hello.

I am a user who uses MSP432P401R, with Energia 0101E0017 (and TI CCS Cloud)
While trying to connect a slave i2c device(IMU) to the board, I discovered that MSP432P401R hangs (specifically, goes to 'sleep mode'(according to CCS Cloud debugging) then 'idle mode' when Wire.endTransmission() is executed. )
*Hang means that it doesn't execute next operations, and stuck in the state until it is manually reset.

I had tried several solutions, such as reinstalling Energia, using most recent build of the MSP432's Wire.h library, and even using external pull-up as suggested in http://forum.43oh.com/topic/3660-i2c-wire-library-endtransmission-hang/.

This appears to occur with and without the i2c device connected
(In arduino, it doesn't occur even when i2c device is disconnected - just the data is fixed to '0')

Thank you for your assistance in advance.

PS. Just for the reference, I attach the source code i am using which is converted for use in Energia
[FinalVer.txt](https://github.com/energia/Energia/files/115204/FinalVer.txt)

engiecat, 2016-02-18:
Update : 
It is revealed that this was due to the difference between the operation of Arduino's endTransmission() and Energia's endTransmission. In case of failed communication, while Arduino moves on to the next operation (returning error code), Energia infinitely 'waits' for the signal, resulting in the hang.

This is probably due to the use of the I2C_masterSendMultiByteFinish() instead of I2C_masterSendMultiByteFinishWithTimeout() in the Wire.h library. (IF the library utilizes MSPWare's driverlib.)

measley, 2016-05-03:
I also ran into the same issue using an I2C motor driver. Curious why the library seemed to work on MSP430 but would freeze up on MSP432. After a decent debug session I found root cause to be endTransmission(). Did you find any fix? I can't see where to update wire library.

closed 2016-08-29 by robertinant