energia/Energia #845: MSP-EXP430FR5739 examples fail to compile with Energia 17

aftisyst, 2016-02-05:
As of Energia 17, MSP-EXP430FR5739 examples are failing to successfully compile with errors as per:

```
Building file: @root/energia-0101E0017/hardware/msp430/cores/msp430/twi_sw.c
Invoking: GNU Compiler
"@root/energia-0101E0017/hardware/tools/msp430/bin/msp430-gcc" -c -mmcu=msp430fr5739 -DF_CPU=16000000L -DENERGIA=17 -DARDUINO=101 -I"@root/energia-0101E0017/hardware/msp430/variants/fraunchpad" -I"@root/energia-0101E0017/hardware/msp430/cores/msp430" -I"@root/energia-0101E0017/hardware/tools/msp430/msp430/include" -Os -ffunction-sections -fdata-sections -g -gdwarf-3 -gstrict-dwarf -Wall -MMD -MP -MF"twi_sw.d" -MT"twi_sw.d" -fmax-errors=10 -o"twi_sw.o"
"
@root/energia-0101E0017/hardware/msp430/cores/msp430/twi_sw.c"
@root/energia-0101E0017/hardware/msp430/cores/msp430/twi_sw.c: In function 'i2c_sw_init':
@root/energia-0101E0017/hardware/msp430/cores/msp430/twi_sw.c:43:16: error: 'TWISDA1' undeclared (first use in this function)
@root/energia-0101E0017/hardware/msp430/cores/msp430/twi_sw.c:43:16: note: each undeclared identifier is reported only once for each function it appears in
@root/energia-0101E0017/hardware/msp430/cores/msp430/twi_sw.c:44:16: error: 'TWISCL1' undeclared (first use in this function)
@root/energia-0101E0017/hardware/msp430/cores/msp430/twi_sw.c: In function 'i2c_sw_start':
@root/energia-0101E0017/hardware/msp430/cores/msp430/twi_sw.c:84:11: error: 'TWISDA1' undeclared (first use in this function)
@root/energia-0101E0017/hardware/msp430/cores/msp430/twi_sw.c:86:11: error: 'TWISCL1' undeclared (first use in this function)
@root/energia-0101E0017/hardware/msp430/cores/msp430/twi_sw.c: In function 'i2c_sw_txByte': @root/energia-0101E0017/hardware/msp430/cores/msp430/twi_sw.c:99:15: error: 'TWISDA1' undeclared (first use in this function)
@root/energia-0101E0017/hardware/msp430/cores/msp430/twi_sw.c:103:13: error: 'TWISCL1' undeclared (first use in this function)
@root/energia-0101E0017/hardware/msp430/cores/msp430/twi_sw.c: In function 'i2c_sw_ack': @root/energia-0101E0017/hardware/msp430/cores/msp430/twi_sw.c:122:12: error: 'TWISCL1' undeclared (first use in this function)
@root/energia-0101E0017/hardware/msp430/cores/msp430/twi_sw.c: In function 'i2c_sw_rxByte':
@root/energia-0101E0017/hardware/msp430/cores/msp430/twi_sw.c:132:12: error: 'TWISDA1' undeclared (first use in this function)
@root/energia-0101E0017/hardware/msp430/cores/msp430/twi_sw.c:136:15: error: 'TWISCL1' undeclared (first use in this function)
@root/energia-0101E0017/hardware/msp430/cores/msp430/twi_sw.c: In function 'i2c_sw_stop':
@root/energia-0101E0017/hardware/msp430/cores/msp430/twi_sw.c:157:11: error: 'TWISDA1' undeclared (first use in this function)
compilation terminated due to -fmax-errors=10.
gmake: *** [twi_sw.o] Error 1
```

This was observed within CCS Cloud IDE using Energia 17.  Reportedly within the Energia IDE v17 itself, the MSP430FR5739 core library fails to build there as well, with the exact same errors.

Thanks!

robertinant, 2016-02-05:
Thanks for the report! I think this snug in with a last minute fix after testing. 
Will look into it.

chemmy1010, 2016-02-19:
I just ran into the same issue with the FraunchPad 5739. Going back to Energia 16 fixes the problem.

robertinant, 2016-02-20:
The fix has been committed. It's a 2 line fix and should be easy to fix in your Energia installation.

https://github.com/energia/Energia/commit/827e338d22f57e03f21ffb5a064c271498d983b6