energia/Energia #847: TM4C1294 - Add Program over Ethernet with LM Flash Utility

rei-vilo, 2016-02-12:
As per http://forum.43oh.com/topic/9350-tm4c1294-program-over-ethernet-with-lm-flash-utility/

miyunluo, 2016-06-07:
Hi,excuse me.

May I ask you for some help?

I have a board EK-TM4C1294XL and I want to download code to the board using LM flash via ethernet.

However I failed and I searched the internet found that a special bootloader is needed.

TI has given an example bootloader and it selects the UART as the port for communicating with the boot loader.I don't know how to change the code to select ethernet.

Can you give me an example?Thank you

entire, 2017-01-17:
I'd be interested in an update on some example code for a flash based ethernet bootloader as well.

Any update on this?