energia/Energia #858: TMS320F28377S Energia sketch compilation failures with Energia 17

aftisyst, 2016-02-29:
With Energia 17 in CCS Cloud, the TMS320F28377S Energia sketches are failing to compile successfully due to a fatal error #1965: "cannot open source file pins_energia.h"

`Building file: @root/energia-0101E0017/hardware/c2000/cores/c2000/HardwareSerial.cpp Invoking: C2000 Compiler "@ccs/tools/compiler/ti-cgt-c2000_6.4.6/bin/cl2000" -v28 -ml -mt --include_path="@root/energia-0101E0017/hardware/c2000/variants/launchpad_f28377S" --include_path="@root/energia-0101E0017/hardware/c2000/cores/c2000" --include_path="@root/energia-0101E0017/hardware/c2000/cores/c2000/F2837xS_common/include" --include_path="@ccs/tools/compiler/ti-cgt-c2000_6.4.6/include" -g --gcc --define=TMS320F28377S --define=F_CPU=200000000 --define=ENERGIA=17 --define=ARDUINO=101 --diag_warning=225 --display_error_number --diag_wrap=off --gen_func_subsections=on --asm_extension=S --preproc_with_compile --preproc_dependency="HardwareSerial.pp" --diag_error=66 --set_error_limit=10 "@root/energia-0101E0017/hardware/c2000/cores/c2000/HardwareSerial.cpp" "@root/energia-0101E0017/hardware/c2000/cores/c2000/Energia.h", line 237: fatal error #1965: cannot open source file "pins_energia.h" >> Compilation failure 1 catastrophic error detected in the compilation of "@root/energia-0101E0017/hardware/c2000/cores/c2000/HardwareSerial.cpp". Compilation terminated. gmake: *** [HardwareSerial.obj] Error 1 Building file: @root/energia-0101E0017/hardware/c2000/cores/c2000/Print.cpp Invoking: C2000 Compiler "@ccs/tools/compiler/ti-cgt-c2000_6.4.6/bin/cl2000" -v28 -ml -mt --include_path="@root/energia-0101E0017/hardware/c2000/variants/launchpad_f28377S" --include_path="@root/energia-0101E0017/hardware/c2000/cores/c2000" --include_path="@root/energia-0101E0017/hardware/c2000/cores/c2000/F2837xS_common/include" --include_path="@ccs/tools/compiler/ti-cgt-c2000_6.4.6/include" -g --gcc --define=TMS320F28377S --define=F_CPU=200000000 --define=ENERGIA=17 --define=ARDUINO=101 --diag_warning=225 --display_error_number --diag_wrap=off --gen_func_subsections=on --asm_extension=S --preproc_with_compile --preproc_dependency="Print.pp" --diag_error=66 --set_error_limit=10 "@root/energia-0101E0017/hardware/c2000/cores/c2000/Print.cpp" "@root/energia-0101E0017/hardware/c2000/cores/c2000/Energia.h", line 237: fatal error #1965: cannot open source file "pins_energia.h" 1 catastrophic error detected in the compilation of "@root/energia-0101E0017/hardware/c2000/cores/c2000/Print.cpp". Compilation terminated. >> Compilation failure gmake: *** [Print.obj] Error 1 Building file: @root/energia-0101E0017/hardware/c2000/cores/c2000/Stream.cpp Invoking: C2000 Compiler "@ccs/tools/compiler/ti-cgt-c2000_6.4.6/bin/cl2000" -v28 -ml -mt --include_path="@root/energia-0101E0017/hardware/c2000/variants/launchpad_f28377S" --include_path="@root/energia-0101E0017/hardware/c2000/cores/c2000" --include_path="@root/energia-0101E0017/hardware/c2000/cores/c2000/F2837xS_common/include" --include_path="@ccs/tools/compiler/ti-cgt-c2000_6.4.6/include" -g --gcc --define=TMS320F28377S --define=F_CPU=200000000 --define=ENERGIA=17 --define=ARDUINO=101 --diag_warning=225 --display_error_number --diag_wrap=off --gen_func_subsections=on --asm_extension=S --preproc_with_compile --preproc_dependency="Stream.pp" --diag_error=66 --set_error_limit=10 "@root/energia-0101E0017/hardware/c2000/cores/c2000/Stream.cpp" "@root/energia-0101E0017/hardware/c2000/cores/c2000/Energia.h", line 237: fatal error #1965: cannot open source file "pins_energia.h" 1 catastrophic error detected in the compilation of "@root/energia-0101E0017/hardware/c2000/cores/c2000/Stream.cpp". Compilation terminated. >> Compilation failure gmake: *** [Stream.obj] Error 1 Building file: @root/energia-0101E0017/hardware/c2000/cores/c2000/TimerAPI.cpp Invoking: C2000 Compiler "@ccs/tools/compiler/ti-cgt-c2000_6.4.6/bin/cl2000" `
etc.

chintanp, 2016-03-09:
I am getting the same error too, when trying to run the examples.

rei-vilo, 2016-03-18:
Which OS please?