energia/Energia #859: Readbear WIFI mini : Wifi.status() does not work and fails the tests.

dx266, 2016-03-08:
Hi I am new with wifi mini have a pb with the OutOfBox example. I figured out that the tests using wifi.status fails. I also tried other example and I have a similar issue with wifi.status but also with wifi.begin and wifi.localIP that all fail.

**TO SUM UP the issues:**
1) with or without client connected I cannot get the followings tests to work. The classes do not change and always return the same:
 while ( status != WL_CONNECTED)...  status always return 7
or
while (WiFi.localIP() == INADDR_NONE)... always return 192.168.1.1

2) I cannot use the wifi.begin() class. It only works with BeginNetwork() class but I did not find any documentation on it.

3) can I get more than one client connected at a time? if yes how to do it ? For now I cannot get more than one.
ps: I have just been updating the firware to the latest version with no change.

Am I doing something wrong ?

Thanks very much for your help

rei-vilo, 2016-03-08:
Have you raised the issue at the RedBear forum?

dx266, 2016-03-08:
Hi Rei,

thanks for your answer. Yes I did but... did not get any feedback so...

What would you recommend?

rei-vilo, 2016-03-08:
Which example are you trying, so I can try and replicate it on the LaunchPad CC3200 and the RedBear CC3200?

dx266, 2016-03-08:
Enclosed you will find a zip file with two of the examples.

I have added the monitor logs:

- for outofbox there is no client connected but the test pass although it
should not. Nevertheless I can see the network and connect a client.
- for the webserver the dots are endless and I cannot see any wifi network.

thanks very much in advance for your help.

rei-vilo, 2016-03-08:
The zip file hasn't been attached. I recommend starting with the examples provided with Energia. Also, make sure your router is configured to accept the connection from the board (e.g. MAC address).

dx266, 2016-03-08:
I have tried the OutOfBox and the WifiWebClient.

**With the OutOfBox** I get this on the monitor but with not client connected:
_Setting up Access Point named: RedBearLab CC3200
Waiting for an ip address

IP Address obtained
    SSID: RedBearLab CC3200
    IP Address: 192.168.1.1
    signal strength (RSSI):0 dBm
    To see this page in action, open a browser to http://192.168.1.1
Starting webserver on port 80
Webserver started!_
  
So the first test fails Wifi.status

**With the WifiWebClient** I have those dots forever: ..........Attempting to connect to Network named: networkname .............................................................................

so the wifi.begin does not work.

I am not using any router. I just use my ipad to connect as a client directly.

rei-vilo, 2016-03-08:
I'll continue the exploration at http://forum.43oh.com/topic/9423-cc3200-wifi-mini-readbear-wifistatus-does-not-work-and-fails-the-tests.

dx266, 2016-03-08:
ok thanks

rei-vilo, 2016-03-08:
The examples work fine on both LaunchPad CC3200 and RedBearLab CC3200 WiFi boards.

closed 2016-03-08 by rei-vilo

rei-vilo, 2016-03-08:
Ok, so you're trying to set the board as an access provider = AP. 

That wasn't clear.

The examples provided with Energia and mentioned in this thread run the board is client mode, and connect it to a server.

AP mode is a whole different story.

Try the example **CreateNetwork**.

rei-vilo, 2016-03-08:
Try the just committed new CreateAPModeWebServer example.

Tested on both LaunchPad CC3200 and RedBearLab CC3200 WiFi boards.

dx266, 2016-03-08:
Oh I see.
So just tried it on a router with wifi.begin() and now yes it works and
pass the tests properly.

I started from the OutOfTheBox example which is in AP mode. Too bad the
wifi.beginNetwork() is not documented as wifi.begin() on the arduino
website.

So one question can I get multiple clients connected to the AP? How ?

thank you very very much for your help.

rei-vilo, 2016-03-08:
There is a limitation that only **ONE** device can be connected.

dx266, 2016-03-08:
Thanks very much Rei for your precious help.