energia/Energia #861: Energia compile error, TMC4123G

qdevery, 2016-03-11:
So I've been trying to compile this test script...

****************Sketch********************
#include <Ads1118.h>
#include <SPI.h>

Ads1118 ADS;

void setup() {
        Serial.begin(9600); // starts serial comm at 9600 baud
	ADS.begin();
        noTone(2);       //ensures no tone coming from ADS BoosterPack
}

void loop() {
        //prints temp in farenheit every 100ms
        Serial.println(ADS.readFarenheit());
        delay(100);
}

...but I always get the errors below. Any ideas on why I'm getting the errors?

*************Error Compiling****************
         ^
sketch_ADS1118_test.cpp:3:9: note: candidates are:
In file included from sketch_ADS1118_test.cpp:1:0:
/Users/xxx/Documents/Energia/libraries/Ads1118/Ads1118.h:16:9: note: Ads1118::Ads1118(int)
         Ads1118(int CS_pin);
         ^
/Users/xxx/Documents/Energia/libraries/Ads1118/Ads1118.h:16:9: note:   candidate expects 1 argument, 0 provided
/Users/xxx/Documents/Energia/libraries/Ads1118/Ads1118.h:13:7: note: Ads1118::Ads1118(const Ads1118&)
 class Ads1118
       ^
/Users/xxx/Documents/Energia/libraries/Ads1118/Ads1118.h:13:7: note:   candidate expects 1 argument, 0 provided
sketch_ADS1118_test.cpp: In function 'void loop()':
sketch_ADS1118_test.cpp:13:28: error: 'class Ads1118' has no member named 'readFarenheit'
         noTone(2);       //ensures no tone coming from ADS BoosterPack
                            ^

rei-vilo, 2016-03-11:
Please post your question at the forum www.43oh.com. You'll receive more answers.

qdevery, 2016-03-11:
I've posted there as well but I thought this would be a better place since the error I'm getting is "error: no matching function for call to 'Ads1118:: Ads1118()'"

Should I be passing some when I instantiate "Ads1118 ADS;"

Ads1118::Ads1118(int CS_pin)
{
    /* Constructor for Ads1118 class. */
    pinMode(CS_pin, OUTPUT);
    _cs = CS_pin;
}

qdevery, 2016-03-11:
NM. Just figured it out. Thanks.

rei-vilo, 2016-03-11:
Answer at http://forum.43oh.com/topic/9416-energia-not-recognizing-ads1118h-as-a-new-library/?p=71370

closed 2016-03-11 by rei-vilo