energia/Energia #865: MSP430 ADC10_B reference setting results in wrong reference selected

barawn, 2016-03-25:
The defines/code for setting the ADC10_B reference in Energia result in the wrong reference selected. Energia attempts to store both the ADC reference select and the reference voltage select in a single word (analog_reference) split via 

`#define REFV_MASK 0x70
#define REF_MASK 0x31`

Obviously this is a little weird already, since these bitmasks overlap. For instance, attempting to set analogReference to INTERNAL1V5 sets analog_reference to 0x11, because

`#define INTERNAL1V5 (ADC10SREF_1 | REFON | REFVSEL_0)`

and ADC10SREF_1 is 0x10, REFON is 0x1, and REFVSEL_0 is 0x0. Then, when analogRead is called, 

`    REFCTL0 |= analog_reference & REF_MASK; // Set reference using masking off the SREF bits. See Energia.h.
    ADC10MCTL0 = channel | (analog_reference & REFV_MASK); // set channel and reference `

which resolves to 

`REFCTL0 |= 0x11 & 00x31;`
or
`REFCTL0 |= 0x11;`

which actually sets the reference to 2.0V, not 1.5V. The fix is probably what the ADC12 conversions do: shift up the ADC10SREF definitions, and then shift them down when they're ORed in. (Although just as a matter of note, shifting up by 8 is probably more efficient than shifting up by 4, as gcc might be able to recognize it as a byte swap and use the byte swap instruction).

barawn, 2016-03-28:
To be clear this occurs with any MSP430 with an ADC10_B, like the MSP430FR5739 present on the "FraunchPad", the  MSP-EXP430FR5739.

StefanSch, 2016-04-08:
you are right, all the different settings of the various ADCs are hard to map.
Reworked the system to have all least the same structure for all ADC variants.
(ADC ref select shifted to upper 8 bits) - this should make the maintenance much easier and reduce setup errors.
Just pushed a first version to branch issue_865.