energia/Energia #866: Exception in thread "main" java.lang.IllegalArgumentException: scrollCount and interval must be greater than 0

SmokinGrunts, 2016-04-01:
Okay, where to start... 
Was working fine a few hours ago, then all of a sudden... No worky. Here's what I get with ./energia:
`Exception in thread "main" java.lang.IllegalArgumentException: scrollCount and interval must be greater than 0
	at processing.app.tools.MenuScroller.<init>(MenuScroller.java:310)
	at processing.app.tools.MenuScroller.<init>(MenuScroller.java:283)
	at processing.app.tools.MenuScroller.setScrollerFor(MenuScroller.java:166)
	at processing.app.Editor.rebuildExamplesMenu(Editor.java:632)
	at processing.app.Editor.buildFileMenu(Editor.java:513)
	at processing.app.Editor.buildMenuBar(Editor.java:475)
	at processing.app.Editor.<init>(Editor.java:206)
	at processing.app.Base.handleOpen(Base.java:778)
	at processing.app.Base.handleOpen(Base.java:743)
	at processing.app.Base.handleNew(Base.java:594)
	at processing.app.Base.<init>(Base.java:331)
	at processing.app.Base.main(Base.java:205)
`

using sudo ./energia works... What happened?! Referencing [this](https://github.com/energia/Energia/issues/558) cuz apparently it shouldn't happen as 

> Robertinant: I will be moving Energia to the new Arduino 1.6 IDE. These issues are no longer applicable when that happens so closing the issue.

Running Ubuntu 15.10,
Intel Core i5-4210U

$uname -a output:
`Linux ss-minnow 4.2.0-34-generic #39-Ubuntu SMP Thu Mar 10 22:13:01 UTC 2016 x86_64 x86_64 x86_64 GNU/Linux`

I normally use CCS/Eclipse, but on occasion I startup Energia to get a (kinda-sometimes) quick check of my serial port. 

Why would this just appear at random?! Whatcha guys need to squash this bug, I'm all ears.

spirilis, 2016-04-01:
It's annoying but either you added libs to the current arch or switched to a different arch that had more built-in libs with examples/etc.  Basically when there are too many examples for all installed libraries for them to reasonably fit in the height of the current screen, Java freaks out and bombs.

Use a larger (higher res) display, remove some of your user-installed libraries... or I would think using CCS's Energia integration would help workaround that (by not requiring the java IDE).

rei-vilo, 2016-04-27:
Update the IDE to release 1.6.7 of Arduino #588 may solve this issue.

hemangjoshi37a, 2017-05-01:
energia stops running at processing.app.Base.main(Base.java:205)