energia/Energia #871: CC3200 — Build Fails with Wire.begin(slaveID)

rei-vilo, 2016-04-10:
This very basic sketch fails on Energia 17.

```
#include "Wire.h"

void setup()
{
    Wire.begin(0x18);
}

void loop()
{

}
```

>
[/Applications/IDE/Energia.app/Contents/Resources/Java/hardware/tools/lm4f/bin/arm-none-eabi-g++, -Os, -nostartfiles, -nostdlib, -Wl,--gc-sections, -T, /Applications/IDE/Energia.app/Contents/Resources/Java/hardware/cc3200/cores/cc3200/cc3200.ld, -Wl,--entry=ResetISR, -mthumb, -mcpu=cortex-m4, -o, /var/folders/hz/wncgspfd5wxgzmly78yvzk_m0000gn/T/build7035399634946287151.tmp/sketch_apr10a.cpp.elf, /var/folders/hz/wncgspfd5wxgzmly78yvzk_m0000gn/T/build7035399634946287151.tmp/sketch_apr10a.cpp.o, /var/folders/hz/wncgspfd5wxgzmly78yvzk_m0000gn/T/build7035399634946287151.tmp/Wire/Wire.cpp.o, /var/folders/hz/wncgspfd5wxgzmly78yvzk_m0000gn/T/build7035399634946287151.tmp/core.a, -L/var/folders/hz/wncgspfd5wxgzmly78yvzk_m0000gn/T/build7035399634946287151.tmp, /Applications/IDE/Energia.app/Contents/Resources/Java/hardware/cc3200/cores/cc3200/driverlib/libdriverlib.a, -lm, -lc, -lgcc, -L/var/folders/hz/wncgspfd5wxgzmly78yvzk_m0000gn/T/build7035399634946287151.tmp, -lm]
sketch_apr10a.cpp.o: In function `setup':
sketch_apr10a.cpp:(.text.setup+0x4): undefined reference to `TwoWire::begin(int)'
collect2: error: ld returned 1 exit status
>

robertinant, 2016-04-11:
I2C (Wire) slave is currently not implemented on the CC3200.

rei-vilo, 2016-04-11:
Weed, I²C slave works on Eenrgia MT, so I'll use Energia MT.

rei-vilo, 2016-04-15:
No luck, I²C slave compiles on Energia MT but doesn't work.

Please implement I²C slave on the CC3200, Energia and Energia MT.