energia/Energia #873: digitalRead() in ISR not working

JasonParmenter, 2016-04-16:
The code below demonstrates that the digitalRead() function does not work inside of an ISR.

`const int encoder0PinA = 11;
const int encoder0PinB = 31;

volatile int pinACount = 0;
volatile int pinBCount = 0;

volatile bool Aflag = false;
volatile bool Bflag = false;

///////////////////////////////////////////////////////////////////////////////////
//////////////////////////////   S E T U P   //////////////////////////////////////
///////////////////////////////////////////////////////////////////////////////////
void setup(){
  
  Serial.begin(9600);
  Serial.println("COUNTER STARTING UP");
 
  pinMode(11,INPUT);
  pinMode(31,INPUT);
  attachInterrupt(encoder0PinA, doEncoderA, CHANGE); 
  attachInterrupt(encoder0PinB, doEncoderB, CHANGE);

}

///////////////////////////////////////////////////////////////////////////////////
////////////////////////////////   L O O P   //////////////////////////////////////
///////////////////////////////////////////////////////////////////////////////////
void loop(){
  
  // SANITY CHECK
  
  Serial.println(digitalRead(11));  // Here to see that the pin does change states 
  Serial.println(digitalRead(31));  // Here to see that the pin does change states
  

  Serial.print("A Flag: ");
  Serial.print(Aflag);         // I never see either of these get set to 1.
  Serial.print("  B Flag: ");
  Serial.println(Bflag);
 ;
  Serial.println();
  delay(500);

}

void doEncoderA(){
  
 Aflag = digitalRead(encoder0PinA);
 

}

void doEncoderB(){

 Bflag = digitalRead(encoder0PinB);
 
}`

fmilburn3, 2016-04-17:
Note that Energia doesn't seem to process CHANGE properly when specified in the attachInterrupt() function either.

abecedarian01, 2016-04-20:
Possibly related to [#860](https://github.com/energia/Energia/issues/860)

closed 2016-08-29 by robertinant