energia/Energia #878: /build/linux/work/hardware/emt/ti/runtime/wiring/cc3200/libraries/WiFi does not exist.

alkhimey, 2016-04-29:
> artium@melon:~/Projects/Energia/build$ (master) ant
> Buildfile: /home/artium/Projects/Energia/build/build.xml
> 
> build:
> 
> revision-check:
> 
> linux-checkos:
> 
> subprojects-build:
> 
> compile:
>   [methods] No changes to PApplet API.
>     [javac] Compiling 34 source files to /home/artium/Projects/Energia/core/bin
>     [javac] 
>     [javac]           WARNING
>     [javac] 
>     [javac] The -source switch defaults to 1.7 in JDK 1.7.
>     [javac] If you specify -target 1.5 you now must also specify -source 1.5.
>     [javac] Ant will implicitly add -source 1.5 for you.  Please change your build file.
>     [javac] warning: [options] bootstrap class path not set in conjunction with -source 1.5
>     [javac] 1 warning
> 
> build:
>       [jar] Building jar: /home/artium/Projects/Energia/core/core.jar
> 
> compile:
>     [mkdir] Created dir: /home/artium/Projects/Energia/app/bin
>     [javac] Compiling 77 source files to /home/artium/Projects/Energia/app/bin
>     [javac] 
>     [javac]           WARNING
>     [javac] 
>     [javac] The -source switch defaults to 1.7 in JDK 1.7.
>     [javac] If you specify -target 1.5 you now must also specify -source 1.5.
>     [javac] Ant will implicitly add -source 1.5 for you.  Please change your build file.
>     [javac] warning: [options] bootstrap class path not set in conjunction with -source 1.5
>     [javac] Note: Some input files use unchecked or unsafe operations.
>     [javac] Note: Recompile with -Xlint:unchecked for details.
>     [javac] 1 warning
>      [copy] Copying 1 file to /home/artium/Projects/Energia/app/bin
>      [copy] Copying /home/artium/Projects/Energia/app/src/processing/app/Resources_ja.properties to /home/artium/Projects/Energia/app/bin/processing/app/Resources_ja.properties
> 
> build:
>       [jar] Building jar: /home/artium/Projects/Energia/app/pde.jar
> 
> check-common-tools:
> 
> download-common-tools-linux:
>      [echo] Downloading common tools
>       [get] Getting: http://energia.nu/files/tools-common-linux64.tar.bz2
>       [get] To: /home/artium/Projects/Energia/build/linux/tools-common.tar.bz2
> 
> unzip-common-tools-linux:
>     [untar] Expanding: /home/artium/Projects/Energia/build/linux/tools-common.tar.bz2 into /home/artium/Projects/Energia/build/linux/work
> 
> linux-build:
>      [copy] Copying 19 files to /home/artium/Projects/Energia/build/linux/work
>      [copy] Copying 2 files to /home/artium/Projects/Energia/build/linux/work
>      [copy] Copying 5 files to /home/artium/Projects/Energia/build/linux/work/lib
> 
> assemble:
>      [copy] Copying 3 files to /home/artium/Projects/Energia/build/linux/work/tools
>      [copy] Copying 2324 files to /home/artium/Projects/Energia/build/linux/work/hardware
>     [mkdir] Created dir: /home/artium/Projects/Energia/build/linux/work/hardware/emt
>      [copy] Copying 103 files to /home/artium/Projects/Energia/build/linux/work/hardware/cc2600emt/libraries
>      [copy] Copied 60 empty directories to 3 empty directories under /home/artium/Projects/Energia/build/linux/work/hardware/cc2600emt/libraries
>      [copy] Copying 95 files to /home/artium/Projects/Energia/build/linux/work/hardware/msp430/libraries
>      [copy] Copied 53 empty directories to 2 empty directories under /home/artium/Projects/Energia/build/linux/work/hardware/msp430/libraries
>      [copy] Copying 96 files to /home/artium/Projects/Energia/build/linux/work/hardware/lm4f/libraries
>      [copy] Copied 53 empty directories to 2 empty directories under /home/artium/Projects/Energia/build/linux/work/hardware/lm4f/libraries
>      [copy] Copying 95 files to /home/artium/Projects/Energia/build/linux/work/hardware/cc3200/libraries
>      [copy] Copied 53 empty directories to 2 empty directories under /home/artium/Projects/Energia/build/linux/work/hardware/cc3200/libraries
>      [copy] Copying 98 files to /home/artium/Projects/Energia/build/linux/work/hardware/msp432/libraries
>      [copy] Copied 56 empty directories to 3 empty directories under /home/artium/Projects/Energia/build/linux/work/hardware/msp432/libraries
>      [copy] Copying 103 files to /home/artium/Projects/Energia/build/linux/work/hardware/cc3200emt/libraries
>      [copy] Copied 60 empty directories to 3 empty directories under /home/artium/Projects/Energia/build/linux/work/hardware/cc3200emt/libraries
> 
> check-closure:
> 
> download-closure:
>      [echo] Downloading closure
>       [get] Getting: http://energia.nu/files/closure-emt-a61.tar.gz
>       [get] To: /home/artium/Projects/Energia/build/linux/closure-emt-a61.tar.gz
> 
> unzip-closure:
>     [untar] Expanding: /home/artium/Projects/Energia/build/linux/closure-emt-a61.tar.gz into /home/artium/Projects/Energia/build/linux/work/hardware/emt
> 
> BUILD FAILED
> /home/artium/Projects/Energia/build/build.xml:82: The following error occurred while executing this line:
> /home/artium/Projects/Energia/build/build.xml:463: The following error occurred while executing this line:
> /home/artium/Projects/Energia/build/build.xml:184: /home/artium/Projects/Energia/build/linux/work/hardware/emt/ti/runtime/wiring/cc3200/libraries/WiFi does not exist.
> 
> Total time: 1 minute 0 seconds

alkhimey, 2016-05-02:
I get this error when trying to build the master branch.

RickKimball, 2016-05-02:
Read the comments on this thread:

https://github.com/energia/Energia/commit/8c25e7c5d6f20df6d4021e1497fb2ac46a35be32

closed 2016-05-02 by robertinant