energia/Energia #879: MSP432 pin mapping is wrong for A9

fmilburn3, 2016-05-05:
See this thread:  http://forum.43oh.com/topic/9614-msp-432-analog-pin-9-mapped-to-analog-pin-8/

A9 is mapped to pin 27 instead of pin 26 on the MSP432.

rei-vilo, 2016-05-05:
Is it 

- the pins map  

[image]

- or the link between MCU pins and LaunchPad pins defined in `pins_energia.h`?

fmilburn3, 2016-05-05:
The problem is in pins_energia.h - in the following lines:
`
static const uint8_t A8 = 27;
static const uint8_t A9 = 27;
static const uint8_t A10 = 6;
`
It should read:
`static const uint8_t A8 = 27;
static const uint8_t A9 = 26;    // change this from 27 to 26
static const uint8_t A10 = 6;
`

rei-vilo, 2016-05-05:
I can't pull the fix because this specific file comes form [emt @ a350116](https://github.com/energia/emt/tree/a350116673c64a5ba71a034b3baaa384b888776e).

>
On branch issue_879
Your branch is up-to-date with 'origin/issue_879'.
Changes not staged for commit:
>	modified:   emt (modified content)
>
no changes added to commit
 (1)
>

rei-vilo, 2016-05-05:
See File [/src/ti/runtime/wiring/msp432/variants/MSP_EXP432P401R/pins_energia.h](https://github.com/rei-vilo/emt/blob/master/src/ti/runtime/wiring/msp432/variants/MSP_EXP432P401R/pins_energia.h) line 56

rei-vilo, 2016-05-05:
See [Fix for A9 on MSP432 #1](https://github.com/energia/emt/issues/1) and pull-request [Fix for A9 on MSP432 #2](https://github.com/energia/emt/pull/2).

Only @robertinant can commit it.

rei-vilo, 2016-05-05:
@fmilburn3 Thank you so much for noticing the bugs and providing the details!

closed 2016-08-29 by robertinant