energia/Energia #880: CC2650 sensortag standalone code for MPU9250.

vinaybk, 2016-05-09:
Hi All,
     I am trying to understand and have full control over MPU9250. Hence i started with TI provided BLE stack + SensortagApp. But then it limited me to experiment with accelerometer experiments. Hence i started developing standalone code for MPU9250 on CC2650. Till now i have not got any success in this direction as well.
         If any of you have standalone code for MPU9250 for CC2650, then it will help me a lot.
Please let me know.

Regards
Vinay

rei-vilo, 2016-05-09:
Regarde to #740

PAk-CatchFire, 2016-09-20:
I have tried with TIVA and MPU9250.
https://github.com/PAk-CatchFire/MPU9250-Tivaware

rei-vilo, 2016-09-20:
Energia 1.6.10E18 no longer supports the CC2650 SensorTag. 

>
CC2650 has been removed from Energia due to the lack of BLE support that could not be enabled because of licensing issues with the BLE stack.
>

closed 2016-09-20 by rei-vilo