energia/Energia #881: ADC_A support + analogReference() bug

battosai30, 2016-05-09:
Hi,

Porting MSP430F5510 to Energia, I discovered several bugs using analogReference().

So first, as ADC_A and B are very similar, it's easy to implement it : just add the good defined(__MSP_430_has_ADC_A__) where defined(__MSP_430_has_ADC_A__) are.

Secondly, when I tried to playu with analogRead() and analogReference() I had strange problems. In fact, I finally discoverd that there are two problems in binary manipulations : 
1)  In energia.h : 
```
#if defined(__MSP430_HAS_ADC10_B__) 
#define DEFAULT (ADC10SREF_0)
#define INTERNAL1V5 ((ADC10SREF_1) | REFON | REFVSEL_0)
#define INTERNAL2V5 ((ADC10SREF_1) | REFON | REFVSEL_2)
#define EXTERNAL (ADC10SREF_2)
#endif
```
The problem here is ADC10SREF and REFVSEL bits have the bit places. Hopefully it's not a problem because with REFVSEL_0 and REFVSEL_2 set the same bit, but I don't think it's a good manner of coding :) Moreover, reference_mode variable, which is set in analogReference() is unint_16, so it's easy to allow the use of internal 2V reference and be clean in code: 
Energia.h : 
```
#if defined(__MSP430_HAS_ADC10_A__)
#define DEFAULT (ADC10SREF_0<<8)
#define INTERNAL1V5 ((ADC10SREF_1<<8) | REFON | REFVSEL_0)
#define INTERNAL2V0 ((ADC10SREF_1<<8) | REFON | REFVSEL_1)
#define INTERNAL2V5 ((ADC10SREF_1<<8) | REFON | REFVSEL_2)
#define EXTERNAL (ADC10SREF_2<<8)
#endif
```
wiring_analog.h, analogRead() : 
`ADC10MCTL0 = channel | ((analog_reference>>8) & REFV_MASK); // set channel and reference `

2) The REFCTL0 register is badly configured espacially this line : 
`REFCTL0 |= (analog_reference & REF_MASK);// | REFMSTR | REFTCOFF; // Set reference using masking off the SREF bits. See Energia.h.`
The problem is the use of an OR. If you set analog ref to 2V5, you can't go back to another reference. Here is my modified routine for ADC_A, tested (for ADC_B I guess it would be the same) :
```
#if defined(__MSP430_HAS_ADC10_A__)
    ADC10CTL0 &= ~ADC10ENC;                 // disable ADC
    ADC10CTL1 = ADC10SSEL_0 | ADC10DIV_4;   // ADC10OSC as ADC10CLK (~5MHz) / 5
    while(REFCTL0 & REFGENBUSY);          // If ref generator busy, WAIT
	REFCTL0 = 0x80 & ~REFTCOFF;
	REFCTL0 |= (analog_reference & REF_MASK);
	while(REFCTL0 & REFGENBUSY);          // If ref generator busy, WAIT
    ADC10MCTL0 = channel | ((analog_reference>>8) & REFV_MASK); // set channel and reference 
    ADC10CTL0 = ADC10ON | ADC10SHT_4;       // turn ADC ON; sample + hold @ 64 × ADC10CLKs
    ADC10CTL1 |= ADC10SHP;                  // ADCCLK = MODOSC; sampling timer
    ADC10CTL2 |= ADC10RES | ADC10SR ;       // 10-bit resolution
    ADC10IFG = 0;                           // Clear Flags
    ADC10IE |= ADC10IE0;                    // Enable interrupts
    __delay_cycles(128);                    // Delay to allow Ref to settle
    ADC10CTL0 |= ADC10ENC | ADC10SC;        // enable ADC and start conversion
    while (ADC10CTL1 & ADC10BUSY) {         // sleep and wait for completion
        __bis_SR_register(CPUOFF + GIE);    // LPM0 with interrupts enabled
    }
    /* POWER: Turn ADC and reference voltage off to conserve power */
    ADC10CTL0 &= ~(ADC10ENC);
    ADC10CTL0 &= ~(ADC10ON);
    REFCTL0 &= ~(REFON);
	REFCTL0 |= REFTCOFF;
#endif 
```
Another point is that internal temp sensor supply has to be set and it was not.

Regards

rei-vilo, 2016-05-09:
Thaï you for reporting the issue.

As the F5510 isn't supported officially, the code for `analogReference()` hasn't been tested against this specific MCU.

battosai30, 2016-05-09:
I understand, but even I have not an MCU with ADC_B available, I'm pretty sure that the bug will happen with this peripheral. 

Moreover, my suggestion about shifting SREF bits will allow 2V internal analog ref to be used.

StefanSch, 2016-05-10:
please have a look at the rework of the ADC on the MSP430 which was done here:
https://github.com/energia/Energia/pull/869
This is already merge to the master branch and will be available with the next release as well.
Can you check if this does already solve your issues. Otherwise it would be good to know to fix this now.

battosai30, 2016-05-10:
Ok I took a look. Good idea to use macros ;)
I haven't tested yet but but seems ok, except one point to be cleared :  REFTCOFF is not managed, will the temp sensor work ?

StefanSch, 2016-05-10:
You are right the REFTCOFF is not used - therefore the TempSensor is never disabled.
So the TempSensor does work but has a little bit more power consumption when capturing another ADC channel. And within Energia it is much easier to handle.