energia/Energia #884: USBSerial.h and low power

battosai30, 2016-05-19:
Hi,

I found a problem when you try to play with USBSerial.h and low power : calling end() doesn't switch off XT2 oscillator, so consumption stays around 200µA.

This can be solved easily : 

```
void USBSerial::end()
{
  USB_disable();
  XT2_Stop();
}
```
Here it's the "direct" way, as in begin() different configurations are managed like this : 

```
    if (USB_PLL_XT == 2){
		#if defined (__MSP430F552x) || defined (__MSP430F550x) || defined(__MSP430F551x)
			P5SEL |= 0x0C;                                      //enable XT2 pins for F5529
		#elif defined (__MSP430F563x_F663x)
			P7SEL |= 0x0C;
		#endif

        XT2_Start(XT2DRIVE_0);                                          //Start the "USB crystal"
    } 
	else {
		#if defined (__MSP430F552x) || defined (__MSP430F550x)
			P5SEL |= 0x10;                                      //enable XT1 pins
		#endif

        XT1_Start(XT1DRIVE_0);                                          //Start the "USB crystal"
    }
```
 it would be a good thing to do the same in end()

StefanSch, 2016-05-20:
Got your point but having just this function means that if someone would like to keep the XT osillator on for some other function - it would not work.
so i added another function which gives the option to switch the oscillator off or leave it on.
Default off should be OK.
void USBSerial::end(uint8_t disableXTAL)
{
  USB_disable();
  if (disableXTAL){
    if (USB_PLL_XT == 2){
		XT2_Stop();
    } else {
		XT1_Stop();
    }
  }
}

battosai30, 2016-05-20:
Good idea :)

StefanSch, 2016-05-20:
fix merged into master with pull request #885

closed 2016-05-20 by StefanSch