energia/Energia #889: 0165E0018 Version Number

rei-vilo, 2016-06-06:
File `/Applications/IDE/Energia-18.app/Contents/Java/lib/version.txt` in charge of the version number contains `1.6.10B2` instead of `0101E0017` for previous versions.

Do you plan to keep the same versioning scheme? 

I strongly suggest to do so. In that case, version number should be **`0165E0018`**.

robertinant, 2016-06-14:
Actually, this version number should be 1.6.10E0018Bx or 1.6.10E0018 for the final release. This is an input when calling ant dist. So I will make sure that I put in the right version when building the beta/release.

closed 2016-06-14 by robertinant

rei-vilo, 2016-06-14:
Latest release of Arduino.CC IDE is 1.6.9.

The `/Applications/ArduinoCC.app/Contents/Java/lib/version.txt` file says `1.6.9` but the parameter passed on to the compiler is `-DARDUINO=10609`. 

Dealing with `10609`is easier than dealing with `1.6.9`.

Up to you for the final choice.