energia/Energia #898: 0165E0018 MSP430 Core Multiple Locations and Possible Conflicts

rei-vilo, 2016-06-15:
Energia suggested to install a package for the MSP430.

[image]

As a result, the MSP430 core folder has multiple locations:

* Common folder `/Applications/IDE/Energia-18.app/Contents/Java/hardware/energia/msp430`
* Specific folder under `/Users/ReiVilo/Library/Energia15/packages/energia/hardware/msp430/1.0.1`

Only one board is available, `MSP430F5529`.

[image]

rei-vilo, 2016-06-15:
Arduino.CC / Genuino experimented with similar solution on release 1.6.6 (AVR platform under specific folder `~/Library/Arduino15/packages/arduino/hardware/avr/1.6.9`) but went back on release 1.6.7. to common folder (AVR platform under common folder `/Applications/Arduino.app/Contents/Java/hardware/arduino/avr`).

I suggest to split the MSP430 into 2 platforms: 

* **MSP430G2** under common folder `/Applications/Energia.app/Contents/Java/hardware/energia/msp430g2`,  as this MCU shouldn't evolve much.
* **MSP430F** under specific folder `~/Library/Energia15/packages/energia/hardware/msp430f/1.0.1`

robertinant, 2016-06-15:
I was working on the MSP430 core and that triggered your beta version not to recognize the builtin one anymore. The issue was that the version number in package_index.json was not correct. I'll get you a new beta.

closed 2016-06-15 by robertinant