energia/Energia #900: MSP432 interrupts on higher pins crash

dronus, 2016-06-18:
If I use 
` attachInterrupt(PUSH2,isr1,RISING);`
it works as intended.

If I however issue 
` attachInterrupt(53,isr1,RISING);` 
in setup(), the MCU crashes, loop() is never executed.

rei-vilo, 2016-06-18:
Is there a pin `53`?

dronus, 2016-06-18:
Yes, Launchpad connector's Pin 53 is MSP432 pin 6 of port 7. It is on the lower row on the Launchpad. 
Actually, lower pin numbers like in the 30's seem to work, as do high numbers as in the 70's, like PUSH1 and PUSH2. I guess only those where tested...  :-(

The attachInterrupt function crashes immediately if the intermediate pin numbers are chosen. No code after is gets executed.

Also I consider it very bad desgin that for MSP432 only connector pin numbers are allowed, while the Launchpad board itself is not decorated with these, it's print only shows the Px_y port / bit code.

rei-vilo, 2016-06-18:
About the pins naming scheme, please refer to [Pin Naming](http://energia.nu/pin-maps/)

>
User’s are encouraged to use the physical pin numbers (1-20, 1-40, or 1-80 depending on the LaunchPad) of the BoosterPack header pins as identifiers in their software.  Because these numbers are consistent between Launchpads this help to ensure the portability of the user’s software.  Some LaunchPads also support the use of GPIO or Port numbers such as “P1_3″.  These names are not recommended for use and will be deprecated in the future.  New LaunchPads that are added to Energia will not support pin names based on GPIO or Port numbers.
>

dronus, 2016-07-14:
Any news on this? I would help by working on the bug myself if anyone could point me how to get rid of all the shipped binary blobs that are linked. Is there any documentaion how those blobs (containing the interrupt configuration etc.) are built?

closed 2016-08-29 by robertinant