energia/Energia #905: Energia adds bad line to sketch*.cpp file after Sketch Verify/Compile

moneypython, 2016-06-25:
I'm using Energia 0101E0017 targeting the C2000 F28027

This causes the build to fail:
> >> Compilation failure
"sketch_jun25a.cpp", line 4: error #34: invalid line number
1 error detected in the compilation of "sketch_jun25a.cpp".

For some reason, Energia adds a problematic line to the sketch*.cpp file:

>   1 #include "Arduino.h"
  2 void setup();
  3 void loop();
  4 #line 0
  5 void setup()
  6 {
  7   // put your setup code here, to run once:
  8 
  9 }
 10 
 11 void loop()
 12 {
 13   // put your main code here, to run repeatedly:
 14 
 15 }
 16

closed 2016-06-25 by moneypython

josele000, 2017-04-21:
i have the same problem any help?

Saraujoc92, 2018-05-27:
If anyone is still having this problem, I found adding an empty line at the beginning of the file removes the compilation error.