energia/Energia #907: Getting Started Guide for Developers

arvindpdmn, 2016-07-13:
I want to port a MSP430G2955-based board to Energia. I have never done similar porting before. So I started looking for some developer guide but could not find any.

I think it will be a useful resource for all Energia developers. I can assist in creating such a guide provided someone gets me started on the right path with some basic information. I can keep adding to the guide as I progress in my own porting task. 

At the moment, I have no idea where to start. I don't even know if I need the submodule `emt` if I have no need for multitasking.

rei-vilo, 2016-07-13:
Energia is derived from Arduino, so the Arduino resources apply for Energia.

Please find the reference at the [Arduino wiki](https://github.com/arduino/Arduino/wiki/_pages).

Another useful place is the [Energia section on the 43oh forum](http://forum.43oh.com/forum/119-energia/).

rei-vilo, 2016-07-13:
More specifically, see 

* #607 Provide Guidelines to Add Support for New MCU
* #214 Energia support for MSP430G2955

closed 2016-07-13 by rei-vilo

rei-vilo, 2016-08-16:
See #912

arvindpdmn, 2016-10-11:
Thanks. Sorry for the late reply. Did not check my notifications for long.