energia/Energia #908: Add documentation on MSP432 binary blob build procedures

dronus, 2016-07-14:
I have filed some bugs the last month on MSP432 and would like to contribute solutions. However, the MSP432 EMT core ships binary blobs, hidden inside trees of a plethora of Energia and third party (TI) contributed files.

Please add some READMEs explaining what files of this mess are actually used, what of them are sources for the binary blobs, and how to rebuild those blobs from the source!

Otherwise it is very hard to contribute to the framework, as almost no code change to the core files triggers any rebuild in the IDE as long as just the ready made blobs are linked.

rei-vilo, 2016-07-14:
What do you call a `blob`?

rei-vilo, 2016-07-14:
See #907.

dronus, 2016-07-14:
With `blob`, I mean binary objects (some files inside the EMT tree) that are linked but from unclear origin (like the source is not provided, or in this case, I guess it is provided, but their source location and procedures how to rebuild it are unclear).

This is in contrast to the Arduino IDE, where the build process relies on sources, so anything inside the core or libraries can be modified, directly affecting the final build.

So for MSP432 we have several pathes containing runtime source (some from Energia, some from TI) and changing those sources doesn't have any effect on builds made with the Energia IDE.

The IDE keeps linking the same static binary code from some pathes and I have no idea how to change this binaries.

rei-vilo, 2016-07-14:
See [Energia / EMT](https://github.com/energia/emt).

dronus, 2016-07-14:
How does https://github.com/energia/emt relate to the files in energia//hardware/emt/ ?   It seems there are many more files in `energia//hardware/emt/`, on the other hand the top level Makefiles are missing. 

The documentation of https://github.com/energia/emt on the other hand does a lot of explanation how to handle different GIT providers, but not so much on how to obtain binaries and how to integrate them with the Energia IDE.

dronus, 2016-07-14:
Maybe we can get through this by a little example. Consider changing the amount of stop bits for Serial.

I edit `energia/hardware/emt/ti/runtime/wiring/HardwareSerial.cpp`, which I GUESS is the responsible file (there are several more implementations of HardwareSerial in energia/hardware/ of course). 
I add the line `uartParams.stopBits = UART_STOP_TWO;` to the implementation of `HardwareSerial::begin`. `UART_STOP_TWO` is defined by TI's `emt/ti/drivers/UART.h`. 

Now what has to be done to build an Energia sketch using this change? 

I browsed energia/hardware for hours. I just guess `energia/hardware/emt/ti/runtime/wiring/msp432/lib/wiring_msp432.m4fg.lib` has to be rebuilt. But I came on that file just by extensively browsing `hardware/`, there is beside the name no clue where it is for, how to change it etc.

closed 2016-08-29 by robertinant