energia/Energia #909: I²C Variable Length Frame Read with Energia

rei-vilo, 2016-07-20:
I'd like to read an I2C variable length frames with Energia. 

Byte 0 gives the length of the frame.

```
0	1	2	3	4	5	6	7	8	9	10	11	12	13	14	15	16	17	18	19	20	21	22	23	24
------------------------------------------------------------------------------------------------------------------------------------
24	1	0	0	0	0	0	247	0	247	0	246	0	246	0	244	0	245	0	245	0	245	0	174	7
24	1	0	1	0	0	0	248	0	248	0	244	0	244	0	245	0	245	0	246	0	246	0	176	7
8	0	0	6	0	0	0	6	0																
12	1	1	0	0	0	0	204	2	186	2	140	1												
8	0	1	0	0	0	8	9	0																
24	1	2	0	0	0	0	51	2	42	2	41	2	36	2	59	2	58	2	54	2	52	2	156	1
8	0	2	255	255	0	0	0	2
```					

In this example, there are 24-, 8-, 12-byte-long frames.

I'm trying...

```	
Wire.requestFrom(I2C_SLAVE, 1, false);
numberBytes = Wire.read();

Wire.requestFrom(I2C_SLAVE, numberBytes);
// read the frame
```	

...but it doesn't work as expected. Second Wire.requestFrom() starts a new frame.
 
Any idea? Thank you!

StefanSch, 2016-07-20:
for which device does this apply?

rei-vilo, 2016-07-20:
The CapTIvate evaluation kit.

StefanSch, 2016-07-21:
Hm, still not clear to me.
So have you ported energia the the FR2633 and see the problem here with the eUSCI or is this just used as the boosterPack but then the question is again what is the device where you did the Energia coding.

rei-vilo, 2016-07-21:
Actually, the CapTIvate board is used as a BoosterPack and runs the standard program generated by the CapTIvate GUI and compiled with CCS, with the `BULK_I2C` option. So the CapTIvate board acts as a slave I²C. 

The problem lies on the MSP432, which is the master I²C, and the default Wire library provided by Energia.

rei-vilo, 2016-07-23:
On the MSP432, calling `Wire.requestFrom(I2C_SLAVE, 48);` gets all the bytes available, _i.e._ sent by the device, up to the `STOP`.

Then, `Wire.available()` gets the exact number of bytes.

I don't know whether the same solution applies for other MCUs.

closed 2016-07-24 by rei-vilo