energia/Energia #911: CC3200 pin map error

fmilburn3, 2016-08-09:
Pin 39 appears to be mapped incorrectly to pin 9 - see this post in 43oh: http://forum.43oh.com/topic/9825-ti-cc3200-relay-on-pin-39-not-working/

william67, 2016-08-10:
Some additional info:
http://forum.43oh.com/topic/9834-grove-base-boosterpack-launchpad-not-working-togetherp/

rei-vilo, 2016-08-11:
I really suggest not to use the **Grove BoosterPack** with the **CC3200 LaunchPad**, as the analog inputs of the CC3200 are limited to 1.8 V. You may fry the MCU if you connect some standard Grive analog devices, which operate at 3.3 V.
 
For the pins of the **CC3200 LaunchPad**, there are so many numbering schemes that Energia focuses solely on the pins numbers mentioned on the pins map. 
 
The same pins map is also available for the **Grove BoosterPack**.
 
On the **CC3200 LaunchPad**, pins `13`, `25`, `26`, and `33` to `40` are not connected by default. The pins map show them in light colour.
 
I'd recommend using the MSP430F5529 LaunchPad for the Grove BoosterPack.
 
Please refer to the references below for all the pins maps.
 
 
References

* [CC3200 LaunchPad pins map](http://embeddedcomputing.weebly.com/launchpad-wifi-cc3200.html)
* [Grove BoosterPack pins map](http://embeddedcomputing.weebly.com/grove-boosterpack.html)
* [MSP430F5529 LaunchPad pins map](http://embeddedcomputing.weebly.com/launchpad-msp430fr5529.html)

rei-vilo, 2016-08-11:
[image]

william67, 2016-08-11:
From seeed.cc

[http://seeed.cc/Grove-Base-BoosterPack-%26amp%3B-Launchpad-not-working-together-t-5980.html](url)

It's working for pin 39 & 38 for now!

closed 2016-08-11 by rei-vilo