energia/Energia #913: SLFSTextFile.ino tests return of SLFS::open incorrectly, so it always thinks it fails

truefishcoder, 2016-08-18:
The return value is checked incorrectly on both calls to SLFS::open in SLFSTextFile.cpp
In SLFSTextFile.ino committed on Feb-24-2015, return value of SLFS::open is tested as:

```
if (SerFlash.open(...) == false)    // incorrect because open() returns SL_FS_OK==0 on success
{
  //handle error
}
```

both cases should be:

```
if (SerFlash.open(...) < SL_FS_OK)   // error results are negative
{
   // handle error
}
```

spirilis, 2016-08-18:
yeah that may have been an artifact of an API change that occurred before SLFS was imported into Energia.  Let me see if I can pull the repo in and look...

spirilis, 2016-08-18:
Posted a commit fixing this for all copies of the library.  Thanks for the bug report!

closed 2016-08-18 by spirilis