energia/Energia #915: Issue import Energia Sketch into Code Composer Studio

NurseBob, 2016-08-21:
I installed the latest Energia release (1.6.10E18) and am now getting the following message with CCS if I try to import a library or sketch:
"No boards or targets found. Ensure that Energia is fully installed on your system, and that corresponding device support has been installed with CCS."
I note there was a similar (now closed) issue in Jan. 2015.

rei-vilo, 2016-08-21:
I guess CCS doesn't support Energia 18 yet.

NurseBob, 2016-08-21:
Bummer, but not the end of the world...

I guess I'll revert to 17 for now if walking through code with CCS...

Thanks.

robertinant, 2016-08-22:
Unfortunately Energia 18 is not yet supported in CCS. The CCS team is working on support for it and it will probably be available in the next major update of CCS. Not sure for when this is scheduled though.

closed 2016-08-22 by robertinant

NurseBob, 2016-08-22:
It's really not a problem; I'm able to do what I need; test in Energia 
17 for almost all that I'm experimenting with, and debug in CCS.  The 
import of E18 projects is not an impediment.  Thanks for the followup 
note. :)

Bob