energia/Energia #916: Enable second serial port on MSP-EXP430FR5739

barawn, 2016-08-23:
The MSP430FR5739 has 2 serial ports, but pins_energia.h only exposes one. The second uses pins 3 and 4, so it's a straightforward fix to just add the AUX serial port definitions.

Most other boards use a define guard checking for __MSP430_HAS_EUSCI_A0__ *or* __MSP430_HAS_EUSCI_A1__, which isn't quite right - you would want to block them separately, in case there's a board with an EUSCI_A0 only. (I don't think there is one currently, but whatever).

Suggested pins_energia.h attached.

[pins_energia.zip](https://github.com/energia/Energia/files/432896/pins_energia.zip)

rei-vilo, 2016-08-23:
See the generic ticket #836 **Provide Other UART, SPI and I²C Ports on LaunchPads** for all additional ports on all boards.

rei-vilo, 2016-08-23:
Could you please open a pull-request for the `pins_energia.h` file?

With release 1.6.10E18, each platform has now a dedicated repository.

For the MSP430, it is [msp430-ng-core (new)](https://github.com/energia/msp430-ng-core) —and [msp430-lg-core (legacy)](https://github.com/energia/msp430-lg-core) for legacy.

barawn, 2016-08-23:
Done. I don't suppose it's possible to update the

barawn, 2016-08-23:
blah, github. Opened pull request [https://github.com/energia/msp430-lg-core/pull/14](url).