energia/Energia #921: How to use sscanf or similar functions in Energia?

nokianhunter, 2016-08-25:
Hi there,

do you have a quick replacement for the sscanf function, that reads formatted data from a string?
I didn't find a reasonable solution and want to use some Arduino code on a MSP430G2553.

[http://www.cplusplus.com/reference/cstdio/sscanf/]
similar issues: 
[https://github.com/esp8266/Arduino/issues/488]
[https://e2e.ti.com/support/microcontrollers/msp430/f/166/t/439318]

Regards
Nokian

rei-vilo, 2016-08-25:
Have you tried `#include stdio.h`?

nokianhunter, 2016-08-25:
Thanks for your answer. I already tried that. I also compared the stdio.h files of the Arduino IDE and Energia. Energia misses some functions.
I suppose there is no other way, than writing that function on my own for my specific use case.

robertinant, 2016-08-26:
This is not supported by the current msp430 gcc compiler. An updated compiler will be released in the next couple weeks that does support this feature. Please note that these functions take up a lot of RAM/FLASH and using them on a MSP430G2553 might result in code no longer fitting in RAM/FLASH

closed 2016-08-26 by robertinant

bduist, 2016-09-23:
Is the compiler updated ?
And how can I get it ?

robertinant, 2016-09-26:
It has not yet been updated. There are some other fixes/features that are being added to the msp430 core before I can move to the new compiler.