energia/Energia #925: Energia still bricks programmers on Ubuntu

dronus, 2016-09-14:
Using Ubuntu 16.04, I need to use the Energia 18 as older versions won't run the IDE at all.

However, Energia 18 again wants to update the Launchpad MSP430FR4133 ezFET, and as the old versions, fails to do so.

The old recovery procedure as of http://www.aglaglobal.com/content/recover-broken-fet-msp430f5529-launchpad-after-ccs-crashes-during-firmware-update for example, won't work anymore as the libmsp430.so provided by DSLite in Energia 18 is 32bit ELF (!). 

-Why does Energia 18 use 32bit libs in the 64bit version?
-Why does Energia still brick the ezFET if an update is needed?
-How to update the Launchpad to an Energia 18 compatible ezFET firmware ?

dronus, 2016-09-15:
Also to mention, it does not only affect the Launchpad MSP430F5529, but all others I used as well (almost all MSP430 based models).

RickKimball, 2016-09-15:
You could install the libmsp430.so from the new 64 bit CCS 6.2.  The added advantage to using the libmsp430.so that comes with CCS is that you don't constantly flip flop between FET versions each time you switch between CCS and energia.

dronus, 2016-09-15:
Does each IDE need an exactly matching ezFET? Is there no backward compatibiliy in either direction?

RickKimball, 2016-09-15:
However, looking at this more .. the dslite executable is 32 bit. So you don't want to do that.
./hardware/tools/DSLite/DebugServer/bin/DSLite: ELF 32-bit LSB executable, Intel 80386, version 1 (SYSV), dynamically linked, interpreter /lib/ld-linux.so.2, for GNU/Linux 2.6.4, stripped

RickKimball, 2016-09-15:
> Does each IDE need an exactly matching ezFET? Is there no backward compatibility in either direction?
No, I don't think so. I've always thrown out whatever comes with Energia and used the one in CCS.  It is updated more frequently in CCS.

dronus, 2016-09-15:
Yes, that is the reason I cant use Energias libmsp430.so for the debricking process by Ubuntus mspdebug which is 64bit of course. So question is, why a 32bit DSLite and libs are shipped in the 64bit build of Energia. And how to get a matching ezFET on your own to the Launchpad, if Energia is incabale of doing so.

dronus, 2016-09-15:
Maybe one can find a matching ezFET libmsp430.so 64bit build that uses the exactly same version as Energias 32bit release does, but that gaves me headaches.

RickKimball, 2016-09-15:
The main reason a new libmsp430.so gets shipped it to support newer chips. The boards that are supported by Energia is pretty limited and the FET already handles them.

RickKimball, 2016-09-15:
This seems to be transitional period.  Even the CCS 6.2 64 bit isn't fully baked. 
https://e2e.ti.com/support/development_tools/code_composer_studio/f/81/p/540462/1969809#1969809

I guess they will slowly move to 64 bit at their own pace.

RickKimball, 2016-09-15:
> Yes, that is the reason I cant use Energias libmsp430.so for the debricking process by Ubuntus mspdebug which is 64bit of course. So question is, why a 32bit DSLite and libs are shipped in the 64bit [...]

You can do two things. 
1.) Use the libmsp430.so and debugger that comes with CCS 6.2 is 64 bit. This works with a 64 bit mspdebug. 
or
2.) Setup a 32 bit VM and build a 32 bit mspdebug you run on 64 bit linux.

dronus, 2016-09-16:
Or 
3.) Use Energia 17 on Ubuntu 14.04 that bricks the board, but is shipping a 64bit libmsp430.so that can be used with Ubuntu's own mspdebug to restore and then update the FET successfully.

It is strange how complicated it is to basically write a predefined binary file into the Launchpads FET Flash... And I really wonder why I passed three versions of incompatible IDE / FETs in just about two months.

RickKimball, 2016-09-16:
1.) CCS
2.) Online Debugger (dev.ti.com)
3.) Energia

None of those are in sync

dronus, 2016-11-02:
And the question is still open, why Energia bricks almost every FET on almost any version of Ubuntu I tried. If it wouldn't, we won't need a discussion about libmsp430.so variants at all..

spirilis, 2016-11-02:
@dronus do you have the udev rules in place?  Linux needs some udev rules to acknowledge the VID/PID of the FET processor's USB BSL mode...

dronus, 2016-11-10:
Just tried if the udev rules would make any difference, resulting in another MSP430FR4133 board bricked :-( . It connects as HID device only now, no ACM devices. mspdebug can't do anything about it either (ezFET recovery failed, Error 73).  I have a whole collection of dead boards now, waiting for some Ubuntu 14.04 mspdebug to revive them, hopefully. 

I wonder if anyone have tested Energia's FET updates, as for me all that ever happend rendered the board dead. And I tried almost any type of board and both Ubuntu 14.04 and 16.04. I wonder why no one seem to have this problems beside me. Boards where sourced by Digikey over the last months.

So the only working procedure to use Energia at all was to use Energia 17 on Ubuntu 14.04 to brick the  board, then restore the FET by using mspdebug from the Ubuntu repository and libmsp430.so as shipped by Energia to restore it, and use only Energia 17 on Ubuntu 14.04 afterwards to program it. 

I think people only using one Ubuntu and owning only one Launchpad would consider Energia totally broken.