energia/Energia #928: Energia IDE 1.6.10E18 crashes on macOS Sierra

rei-vilo, 2016-09-21:
Energia IDE 1.6.10E18 crashes on macOS Sierra, for most of the boards.

This issue also occurs with the Arduino 1.6.11 IDE, see https://github.com/arduino/Arduino/issues/5397 

Additionally, some paths seem to be hard coded, like `	/Users/robertinant/github/arduino-builder/src/arduino.cc/builder/phases/core_builder.go:55 +0x20f fp=0xc208125800 sp=0xc2081256e0`.

>
/Applications/IDE/Energia.app/Contents/Java/arduino-builder -compile -logger=machine -hardware "/Applications/IDE/Energia.app/Contents/Java/hardware" -hardware "/Users/ReiVilo/Library/Energia15/packages" -tools "/Applications/IDE/Energia.app/Contents/Java/tools-builder" -tools "/Applications/IDE/Energia.app/Contents/Java/hardware/tools/msp430" -tools "/Users/ReiVilo/Library/Energia15/packages" -built-in-libraries "/Applications/IDE/Energia.app/Contents/Java/libraries" -libraries "/Users/ReiVilo/Documents/Projets/Energia/libraries" -fqbn=energia:msp430:MSP-EXP430FR4133LP -ide-version=10610 -build-path "/var/folders/hz/wncgspfd5wxgzmly78yvzk_m0000gn/T/build9548c7de2eb0fa195db29d7f1b03d466.tmp" -warnings=none -prefs=build.warn_data_percentage=75 -verbose "/Applications/IDE/Energia.app/Contents/Java/examples/01.Basics/Blink/Blink.ino"

failed MSpanList_Insert 0x3109b0 0xb9a24b98c8d 0x0
fatal error: MSpanList_Insert

runtime stack:
runtime.MSpanList_Insert(0x2c8318, 0x3109b0)
	/usr/local/go/src/runtime/mheap.c:692 +0x8f
runtime.MHeap_Alloc(0x2c7d60, 0x6, 0x10000000034, 0xf989)
	/usr/local/go/src/runtime/mheap.c:240 +0x66
runtime.MCentral_CacheSpan(0x2d2118, 0x1)
	/usr/local/go/src/runtime/mcentral.c:85 +0x167
runtime.MCache_Refill(0x309000, 0xc200000034, 0xc208012000)
	/usr/local/go/src/runtime/mcache.c:90 +0xa0

goroutine 1 [running]:
runtime.switchtoM()
	/usr/local/go/src/runtime/asm_amd64.s:198 fp=0xc208124de0 sp=0xc208124dd8
runtime.mallocgc(0x2200, 0x1a0b40, 0xc200000000, 0xc2081a7600)
	/usr/local/go/src/runtime/malloc.go:178 +0x849 fp=0xc208124e90 sp=0xc208124de0
runtime.newarray(0x1a0b40, 0x20, 0xa501)
	/usr/local/go/src/runtime/malloc.go:365 +0xc1 fp=0xc208124ec8 sp=0xc208124e90
runtime.hashGrow(0x1b5060, 0xc20810b440)
	/usr/local/go/src/runtime/hashmap.go:744 +0x86 fp=0xc208124ef8 sp=0xc208124ec8
runtime.mapassign1(0x1b5060, 0xc20810b440, 0xc208125010, 0xc208125000)
	/usr/local/go/src/runtime/hashmap.go:456 +0x568 fp=0xc208124f98 sp=0xc208124ef8
arduino.cc/builder/props.PropertiesMap.Merge(0xc20810b440, 0xc2081250c0, 0x1, 0x1, 0x0)
	/Users/robertinant/github/arduino-builder/src/arduino.cc/builder/props/properties.go:162 +0x153 fp=0xc208125090 sp=0xc208124f98
arduino.cc/builder/props.PropertiesMap.Clone(0xc20803dad0, 0x10cb8)
	/Users/robertinant/github/arduino-builder/src/arduino.cc/builder/props/properties.go:170 +0xad fp=0xc2081250e8 sp=0xc208125090
arduino.cc/builder/builder_utils.compileFileWithRecipe(0xc2080a64b0, 0x50, 0xc2081ca7e0, 0x61, 0xc2080381e0, 0x5f, 0xc20803dad0, 0xc20801f320, 0x2, 0x2, ...)
	/Users/robertinant/github/arduino-builder/src/arduino.cc/builder/builder_utils/utils.go:131 +0x66 fp=0xc208125240 sp=0xc2081250e8
arduino.cc/builder/builder_utils.compileFilesWithRecipe(0xc20807f700, 0x9, 0x10, 0xc2080a64b0, 0x50, 0xc20807e100, 0xd, 0x10, 0xc2080381e0, 0x5f, ...)
	/Users/robertinant/github/arduino-builder/src/arduino.cc/builder/builder_utils/utils.go:120 +0x1c3 fp=0xc208125378 sp=0xc208125240
arduino.cc/builder/builder_utils.compileFilesWithExtensionWithRecipe(0xc208125610, 0x0, 0x0, 0xc2080a64b0, 0x50, 0xc2080d3601, 0xc2080381e0, 0x5f, 0xc20803dad0, 0xc20801f320, ...)
	/Users/robertinant/github/arduino-builder/src/arduino.cc/builder/builder_utils/utils.go:87 +0x248 fp=0xc208125478 sp=0xc208125378
arduino.cc/builder/builder_utils.CompileFiles(0xc208125610, 0x0, 0x0, 0xc2080a64b0, 0x50, 0x1, 0xc2080381e0, 0x5f, 0xc20803dad0, 0xc20801f320, ...)
	/Users/robertinant/github/arduino-builder/src/arduino.cc/builder/builder_utils/utils.go:71 +0x31e fp=0xc208125560 sp=0xc208125478
arduino.cc/builder/phases.compileCore(0xc2080381e0, 0x5f, 0xc20803dad0, 0x1, 0x7fff5fbffcc7, 0x4, 0x31adc8, 0x2c19a8, 0x0, 0x0, ...)
	/Users/robertinant/github/arduino-builder/src/arduino.cc/builder/phases/core_builder.go:87 +0x66a fp=0xc2081256e0 sp=0xc208125560
arduino.cc/builder/phases.(*CoreBuilder).Run(0x2c19a8, 0xc208068000, 0x0, 0x0)
	/Users/robertinant/github/arduino-builder/src/arduino.cc/builder/phases/core_builder.go:55 +0x20f fp=0xc208125800 sp=0xc2081256e0
arduino.cc/builder.runCommands(0xc208068000, 0xc208125a98, 0x22, 0x22, 0x1, 0x0, 0x0)
	/Users/robertinant/github/arduino-builder/src/arduino.cc/builder/builder.go:188 +0x139 fp=0xc208125890 sp=0xc208125800
arduino.cc/builder.(*Builder).Run(0xc208125ce0, 0xc208068000, 0x0, 0x0)
	/Users/robertinant/github/arduino-builder/src/arduino.cc/builder/builder.go:123 +0x1227 fp=0xc208125cc0 sp=0xc208125890
arduino.cc/builder.RunBuilder(0xc208068000, 0x0, 0x0)
	/Users/robertinant/github/arduino-builder/src/arduino.cc/builder/builder.go:219 +0x49 fp=0xc208125ce8 sp=0xc208125cc0
main.main()
	/Users/robertinant/github/arduino-builder/src/arduino.cc/arduino-builder/main.go:316 +0x171b fp=0xc208125f98 sp=0xc208125ce8
runtime.main()
	/usr/local/go/src/runtime/proc.go:63 +0xf3 fp=0xc208125fe0 sp=0xc208125f98
runtime.goexit()
	/usr/local/go/src/runtime/asm_amd64.s:2232 +0x1 fp=0xc208125fe8 sp=0xc208125fe0
arduino-builder returned 2
Error compiling for board MSP-EXP430FR4133LP.

>

rei-vilo, 2016-09-21:
See http://forum.43oh.com/topic/9911-energia-ide-1610e18-crashes-on-macos-sierra/

rei-vilo, 2016-09-21:
As a temporary solution, replace the `arduino-builder` utility of Energia (version 1.3.18) by the new Arduino one (version 1.3.21)

- Download the [Arduino 1.6.12 IDE](https://www.arduino.cc/en/Main/Software), 
- Look for the `arduino-builder` utility, under `/Applications/Arduino.app/Contents/Java/arduino-builder` on macOS,
- Copy and paste it into `/Applications/Energia.app/Contents/Java/arduino-builder` on macOS.

robertinant, 2016-09-21:
Note that the arduino builder included in Energia is slightly modified. Please see http://forum.43oh.com/topic/9911-energia-ide-1610e18-crashes-on-macos-sierra/?p=74625 for details. It has a new build of arduino-builder attached which is the same version included in Arduino 1.6.12. The original arduino builder included in the Arduino IDE will not be able to compile EMT targets.

rei-vilo, 2016-09-21:
I've tested with CC3200 and MSP430 mostly, not MSP432 EMT yet. Thank you for the tip.

rei-vilo, 2016-09-21:
Please use the file provided by Robert. This was a very temporary solution!

http://forum.43oh.com/index.php?app=core&module=attach&section=attach&attach_id=9229

robertinant, 2016-09-21:
Did that arduino-builder work for you for both EMT and none-EMT targets?

rei-vilo, 2016-09-21:
Compiling the `MultiBlink` example against the MSP432 with the `arduino-builder` from Arduino 1.6.12 succeeded.

> Sketch uses 23,944 bytes (9%) of program storage space. Maximum is 262,144 bytes.

robertinant, 2016-09-21:
Perfect. Thanks for reporting back! Unfortunately arduino-builder can not be updated outside the IDE using the board manager so I will have to create a new release of Energia. Will do so in the next couple of days.

cameronus, 2016-12-01:
@robertinant I tried this quick fix, but it didn't work for me. After I clicked Upload, the arduino-builder process began consuming my CPU resources and hovered at around 60%. I have verbose debug on, and the upload debugging messages froze. Even after closing Energia, I had to manually terminate the rogue arduino-builder process. I have repeated this multiple times. If you have any other fix I would be extremely grateful, because this issue is persistent across many boards I have tested it on.

Thanks,
Cameron

rei-vilo, 2016-12-02:
I'm also facing the issue with `arduino-builder`. 

It takes 60% of CPU and I have no alternative but to kill it.

rei-vilo, 2016-12-04:
Again.

[image]

MatthewWest, 2016-12-20:
@rei-vilo You mentioned that the file in the forums is a 

> very temporary solution

Is there a permanent solution or a timeline on when there will be one?

cameronus, 2016-12-20:
@rei-vilo @MatthewWest @robertinant I also am still searching for a solution to this problem. Can you confirm that it will be in the next release or that there is another solution in the works?

rei-vilo, 2017-02-12:
Still the same issue. Here the statistics of the `arduino-builder` freezing.

> Sampling process 14614 for 3 seconds with 1 millisecond of run time between samples
> Couldn't find _sigtramp symbol in expected dylibs
> Sampling completed, processing symbols...
> Analysis of sampling arduino-builder (pid 14614) every 1 millisecond
> Process:         arduino-builder [14614]
> Path:            /Applications/IDE/Energia.app/Contents/Java/arduino-builder
> Load Address:    0x1000
> Identifier:      arduino-builder
> Version:         ???
> Code Type:       X86-64
> Parent Process:  Energia [13564]
> 
> Date/Time:       2017-02-12 12:10:35.851 +0100
> Launch Time:     2017-02-12 12:10:20.910 +0100
> OS Version:      Mac OS X 10.12.3 (16D32)
> Report Version:  7
> Analysis Tool:   /usr/bin/sample
> ----
> 
> Call graph:
>     2855 Thread_879404: Main Thread   DispatchQueue_<multiple>
>     + 2855 runtime.sigtrampgo  (in arduino-builder) + 204  [0x3ed5c]  signal_darwin.go:68
>     2855 Thread_879407
>     + 2851 runtime.usleep  (in arduino-builder) + 44  [0x5af9c]  sys_darwin_amd64.s:294
>     + 1 ???  (in <unknown binary>)  [0x7469622d343620]
>     + ! 1 nanotime  (in arduino-builder) + 18  [0x5ad32]  sys_darwin_amd64.s:126
>     + 1 runtime.netpoll  (in arduino-builder) + 40  [0x29e38]  netpoll_kqueue.go:65
>     + 1 runtime.retake  (in arduino-builder) + 541  [0x38a0d]  proc.go:3590
>     + 1 runtime.retake  (in arduino-builder) + 623  [0x38a5f]  proc.go:3598
>     2855 Thread_879408
>     + 2855 runtime.semasleep1  (in arduino-builder) + 223  [0x2ad1f]  os1_darwin.go:423
>     +   2855 runtime.mach_semaphore_wait  (in arduino-builder) + 11  [0x5b0cb]  sys_darwin_amd64.s:416
>     2855 Thread_879409
>     + 2855 runtime.semasleep1  (in arduino-builder) + 223  [0x2ad1f]  os1_darwin.go:423
>     +   2855 runtime.mach_semaphore_wait  (in arduino-builder) + 11  [0x5b0cb]  sys_darwin_amd64.s:416
>     2855 Thread_879410
>     + 2855 runtime.semasleep1  (in arduino-builder) + 223  [0x2ad1f]  os1_darwin.go:423
>     +   2855 runtime.mach_semaphore_wait  (in arduino-builder) + 11  [0x5b0cb]  sys_darwin_amd64.s:416
>     2855 Thread_879411
>     + 2855 runtime.semasleep1  (in arduino-builder) + 223  [0x2ad1f]  os1_darwin.go:423
>     +   2855 runtime.mach_semaphore_wait  (in arduino-builder) + 11  [0x5b0cb]  sys_darwin_amd64.s:416
>     2855 Thread_879412
>     + 2855 runtime.semasleep1  (in arduino-builder) + 223  [0x2ad1f]  os1_darwin.go:423
>     +   2855 runtime.mach_semaphore_wait  (in arduino-builder) + 11  [0x5b0cb]  sys_darwin_amd64.s:416
>     2855 Thread_879413
>     + 2855 runtime.semasleep1  (in arduino-builder) + 223  [0x2ad1f]  os1_darwin.go:423
>     +   2855 runtime.mach_semaphore_wait  (in arduino-builder) + 11  [0x5b0cb]  sys_darwin_amd64.s:416
>     2855 Thread_879414
>     + 2855 runtime.semasleep1  (in arduino-builder) + 223  [0x2ad1f]  os1_darwin.go:423
>     +   2855 runtime.mach_semaphore_wait  (in arduino-builder) + 11  [0x5b0cb]  sys_darwin_amd64.s:416
>     2855 Thread_879427
>       2855 runtime.semasleep1  (in arduino-builder) + 126  [0x2acbe]  os1_darwin.go:412
>         2855 runtime.mach_semaphore_timedwait  (in arduino-builder) + 19  [0x5b0e3]  sys_darwin_amd64.s:426
> 
> Total number in stack (recursive counted multiple, when >=5):
>         7       runtime.mach_semaphore_wait  (in arduino-builder) + 0  [0x5b0c0]  sys_darwin_amd64.s:416
>         7       runtime.semasleep1  (in arduino-builder) + 223  [0x2ad1f]  os1_darwin.go:423
> 
> Sort by top of stack, same collapsed (when >= 5):
>         runtime.mach_semaphore_wait  (in arduino-builder)        19985
>         runtime.mach_semaphore_timedwait  (in arduino-builder)        2855
>         runtime.sigtrampgo  (in arduino-builder)        2855
>         runtime.usleep  (in arduino-builder)        2851
> 
> Binary Images:
>             0x1000 -           0x2ee8b9 +arduino-builder (???) /Applications/IDE/Energia.app/Contents/Java/arduino-builder
>          0x5ece000 -          0x5f0b267  dyld (0.0 - ???) <947FC440-80F9-32F7-A773-6FC418FE1AB7> /usr/lib/dyld
> Sample analysis of process 14614 written to file /dev/stdout
>

rei-vilo, 2017-02-12:
Building a second time often ends in a `panic runtime error`.

> Detecting libraries used...
> "/Users/ReiVilo/Library/Energia15/packages/energia/tools/arm-none-eabi-gcc/4.8.4-20140725/bin/arm-none-eabi-g++" -mcpu=cortex-m4 -mthumb -mfloat-abi=hard -mfpu=fpv4-sp-d16 -mabi=aapcs -g -Dxdc_target_types__=gnu/targets/arm/std.h -Dxdc_target_name__=M4F -Dxdc_cfg__xheader__="configPkg/package/cfg/energia_pm4fg.h" -DTARGET_IS_MSP432P4XX -DBOARD_MSP_EXP432P401R -Dxdc__nolocalstring=1 -D__MSP432P401R__ -DCORE_VERSION=380 -c -g -Os -w -ffunction-sections -fdata-sections -fno-threadsafe-statics -fno-rtti -fno-exceptions  -w -x c++ -E -CC -mcpu=cortex-m4 -mthumb -DF_CPU=48000000L -DARDUINO=10610 -DENERGIA=10610 -DENERGIA_MSP_EXP432P401R -DENERGIA_ARCH_MSP432 "-I/Users/ReiVilo/Library/Energia15/packages/energia/hardware/msp432/3.8.0/cores/msp432/ti/runtime/wiring/" "-I/Users/ReiVilo/Library/Energia15/packages/energia/hardware/msp432/3.8.0/cores/msp432/ti/runtime/wiring/msp432" "-I/Users/ReiVilo/Library/Energia15/packages/energia/hardware/msp432/3.8.0/system/driverlib/MSP432P4xx" "-I/Users/ReiVilo/Library/Energia15/packages/energia/hardware/msp432/3.8.0/system/driverlib" "-I/Users/ReiVilo/Library/Energia15/packages/energia/hardware/msp432/3.8.0/system/inc" "-I/Users/ReiVilo/Library/Energia15/packages/energia/hardware/msp432/3.8.0/system/inc/CMSIS/" "-I/Users/ReiVilo/Library/Energia15/packages/energia/hardware/msp432/3.8.0/system" "-I/Users/ReiVilo/Library/Energia15/packages/energia/hardware/msp432/3.8.0/cores/msp432/ti/runtime/wiring/" "-I/Users/ReiVilo/Library/Energia15/packages/energia/hardware/msp432/3.8.0/cores/msp432/ti/runtime/wiring/msp432" "-I/Users/ReiVilo/Library/Energia15/packages/energia/hardware/msp432/3.8.0/system/driverlib/MSP432P4xx" "-I/Users/ReiVilo/Library/Energia15/packages/energia/hardware/msp432/3.8.0/system/driverlib" "-I/Users/ReiVilo/Library/Energia15/packages/energia/hardware/msp432/3.8.0/system/inc" "-I/Users/ReiVilo/Library/Energia15/packages/energia/hardware/msp432/3.8.0/system/inc/CMSIS/" "-I/Users/ReiVilo/Library/Energia15/packages/energia/hardware/msp432/3.8.0/system" "-I/Users/ReiVilo/Library/Energia15/packages/energia/hardware/msp432/3.8.0/cores/msp432" "-I/Users/ReiVilo/Library/Energia15/packages/energia/hardware/msp432/3.8.0/variants/MSP_EXP432P401R" "/var/folders/px/cyfvtr757lqg0yp_cv9j79jh0000gn/T/build3907647dabd102eb118910a458cffd53.tmp/sketch/Test.ino.cpp" -o "/dev/null"
> Using cached library dependencies for file: /var/folders/px/cyfvtr757lqg0yp_cv9j79jh0000gn/T/build3907647dabd102eb118910a458cffd53.tmp/sketch/LocalLibrary.cpp
> Using cached library dependencies for file: /var/folders/px/cyfvtr757lqg0yp_cv9j79jh0000gn/T/build3907647dabd102eb118910a458cffd53.tmp/sketch/Timer.cpp
> panic: runtime error: index out of range
> 
> goroutine 1 [running]:
> panic(0x1c8940, 0xc820014050)
> 	/usr/local/go/src/runtime/panic.go:481 +0x3e6
> arduino.cc/builder.findIncludesUntilDone(0xc82007d180, 0xc82047be60, 0x16e4c0, 0xc820167ef0, 0xc820192682, 0x8, 0x0, 0x0)
> 	/Users/robertinant/github/arduino-builder/src/arduino.cc/builder/container_find_includes.go:321 +0x13f1
> arduino.cc/builder.(*ContainerFindIncludes).Run(0x31d740, 0xc82007d180, 0x0, 0x0)
> 	/Users/robertinant/github/arduino-builder/src/arduino.cc/builder/container_find_includes.go:149 +0x7e8
> arduino.cc/builder.runCommands(0xc82007d180, 0xc8200ad998, 0x22, 0x22, 0x1, 0x0, 0x0)
> 	/Users/robertinant/github/arduino-builder/src/arduino.cc/builder/builder.go:189 +0x12b
> arduino.cc/builder.(*Builder).Run(0xc8200adbe0, 0xc82007d180, 0x0, 0x0)
> 	/Users/robertinant/github/arduino-builder/src/arduino.cc/builder/builder.go:124 +0x113e
> arduino.cc/builder.RunBuilder(0xc82007d180, 0x0, 0x0)
> 	/Users/robertinant/github/arduino-builder/src/arduino.cc/builder/builder.go:220 +0x39
> main.main()
> 	/Users/robertinant/github/arduino-builder/src/arduino.cc/arduino-builder/main.go:316 +0x1940
> arduino-builder returned 2
> Error compiling for board LaunchPad w/ msp432 EMT (48MHz).
>

closed 2017-03-28 by rei-vilo