energia/Energia #929: Energia 1.6.10E18 won't compile for MSP430

VinceC7, 2016-09-22:
I'm getting a compile error for MSP-EXP430G2553LP:

msp430-g++: error: TI: No such file or directory
msp430-g++: error: boards\energia-1.6.10E18\hardware\tools\msp430/include: No such file or directory

I can compile the programs in E17 thou?

rei-vilo, 2016-09-22:
Which OS please?

VinceC7, 2016-09-22:
Yes, on Windows 8.1, and the path has no spaces.  Strange the error gives the include path as a forward bracket /  ?????

robertinant, 2016-09-22:
Can you copy paste the full path in which you have installed Energia?
Can you also try to put Energia in c:\energia

VinceC7, 2016-09-23:
OK, I got Energia compiling by copying everything to C:\.   Very strange that Energia works for compiling the CC3200 launchpad in the old location.  Also fixed the boards.txt so that the hex file would upload: https://github.com/energia/msp430-lg-core/issues/12

Thanks.
    Vince

closed 2016-09-23 by VinceC7

robertinant, 2016-09-23:
The msp430-gcc has some quirks when it comes to path names. Spaces, to long, etc. This will be fixed when I move to the new msp430-elf-gcc toolchain.

NTHANHCHINH, 2017-02-26:
I got  error compiling for board MSP-EXP430G2553LP on a NeoNextion example "button" (and many example else). while Arduio compiled them OK. 
Please help!

rei-vilo, 2017-02-26:
Please share your code and ask your question on the 43oh forum at http://forum.43oh.com/forum/119-energia/

hemangjoshi37a, 2017-05-01:
I am using energia in Linux Mint. In which direcctory should I run the programm??

Thax in advance.

robertinant, 2017-05-01:
@hemangjoshi37a make sure that you install it in a path with no spaces if you are using an MSP430 LaunchPad. Also, if you are using an ARM based board such as MSP432/CC3200 and are on 64-bit Linux then make sure that you install the 32-bit compatibility libraries for your Linux distribution.

hemangjoshi37a, 2017-05-03:
@robertinant Thax bro for reply.
There is no spaces in the address of the energia library folder : the address is "/media/hemang/586E740B6E73DFE4/msp/energia-1.6.10E18/libraries"

I am using only msp430g2553 not any other controller.

Can you suggest any website for downloading 32bit libraries?? 32bit libraries are not available on official page. Infact they only give libraries without telling if it is 32bit or 64bit. 

I hope to see you soon. thax.

hemangjoshi37a, 2017-05-03:
Error in Energia error message:

fatal error: avr/io.h: No such file or directory
compilation terminated.
exit status 1
Error compiling for board MSP-EXP430G2553LP.

robertinant, 2017-05-03:
Google "linux mint install 32 bit libraries". There are quite a few forum posts that have suggestions.

As for the avr/io.h error. You are compiling either a library or a Sketch that was written for an Arduino board.

For future questions please use the Energia forum @ http://forum.43oh.com/forum/119-energia/

hemangjoshi37a, 2017-05-04:
okay. No problem...

hemangjoshi37a, 2017-05-04:
I have used Energia/libraries/SD/examples/Datalogger/Datalogger.ino this llibrary which is made for Energia and it is showing error. So there is someting wrong with my system.

StefanSch, 2017-05-04:
SD lib:
where did you get the library - you mentioned it is made for Energia.
I am not aware that it is already ported to Energia - except the version on my PC  :-)

What error message? 
Do you get which device ?

Be more specific if you would like to get good answers.

rei-vilo, 2017-05-04:
@hemangjoshi37a 

And please use the dedicated forum http://forum.43oh.com/forum/119-energia/